linux-nvme/nvme-cli #1628: 2.1: build fails

kloczek, 2022-08-01:
Looks like something is wrong ..
```console
[tkloczko@devel-g2v x86_64-redhat-linux-gnu]$ ninja
[5/47] Compiling C object nvme.p/wrapper.c.o
FAILED: nvme.p/wrapper.c.o
/usr/bin/gcc -Invme.p -I. -I.. -Iccan -I../ccan -I/usr/include/uuid -I/usr/include/json-c -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -O0 -fomit-frame-pointer -D_GNU_SOURCE -include config.h -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -MD -MQ nvme.p/wrapper.c.o -MF nvme.p/wrapper.c.o.d -o nvme.p/wrapper.c.o -c ../wrapper.c
../wrapper.c:12:58: warning: ‘enum nvme_version’ declared inside parameter list will not be visible outside of this definition or declaration
   12 | const char * __attribute__((weak)) nvme_get_version(enum nvme_version type)
      |                                                          ^~~~~~~~~~~~
../wrapper.c:12:71: error: parameter 1 (‘type’) has incomplete type
   12 | const char * __attribute__((weak)) nvme_get_version(enum nvme_version type)
      |                                                     ~~~~~~~~~~~~~~~~~~^~~~
../wrapper.c: In function ‘nvme_get_version’:
../wrapper.c:19:44: error: type of formal parameter 1 is incomplete
   19 |                 return libnvme_get_version(type);
      |                                            ^~~~
[8/47] Compiling C object nvme.p/fabrics.c.o
FAILED: nvme.p/fabrics.c.o
/usr/bin/gcc -Invme.p -I. -I.. -Iccan -I../ccan -I/usr/include/uuid -I/usr/include/json-c -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -O0 -fomit-frame-pointer -D_GNU_SOURCE -include config.h -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -MD -MQ nvme.p/fabrics.c.o -MF nvme.p/fabrics.c.o.d -o nvme.p/fabrics.c.o -c ../fabrics.c
../fabrics.c: In function ‘__discover’:
../fabrics.c:411:45: error: ‘ENVME_CONNECT_ALREADY’ undeclared (first use in this function); did you mean ‘ENVME_CONNECT_READ’?
  411 |                         } else if (errno == ENVME_CONNECT_ALREADY && !quiet) {
      |                                             ^~~~~~~~~~~~~~~~~~~~~
      |                                             ENVME_CONNECT_READ
../fabrics.c:411:45: note: each undeclared identifier is reported only once for each function it appears in
[9/47] Compiling C object nvme.p/plugin.c.o
FAILED: nvme.p/plugin.c.o
/usr/bin/gcc -Invme.p -I. -I.. -Iccan -I../ccan -I/usr/include/uuid -I/usr/include/json-c -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -O0 -fomit-frame-pointer -D_GNU_SOURCE -include config.h -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -MD -MQ nvme.p/plugin.c.o -MF nvme.p/plugin.c.o.d -o nvme.p/plugin.c.o -c ../plugin.c
../plugin.c: In function ‘version’:
../plugin.c:24:17: warning: implicit declaration of function ‘nvme_get_version’; did you mean ‘nvme_get_nsid’? [-Wimplicit-function-declaration]
   24 |                 nvme_get_version(NVME_VERSION_PROJECT),
      |                 ^~~~~~~~~~~~~~~~
      |                 nvme_get_nsid
../plugin.c:24:34: error: ‘NVME_VERSION_PROJECT’ undeclared (first use in this function)
   24 |                 nvme_get_version(NVME_VERSION_PROJECT),
      |                                  ^~~~~~~~~~~~~~~~~~~~
../plugin.c:24:34: note: each undeclared identifier is reported only once for each function it appears in
../plugin.c:25:34: error: ‘NVME_VERSION_GIT’ undeclared (first use in this function); did you mean ‘NVME_VERSION’?
   25 |                 nvme_get_version(NVME_VERSION_GIT));
      |                                  ^~~~~~~~~~~~~~~~
      |                                  NVME_VERSION
[13/47] Compiling C object nvme.p/plugins_zns_zns.c.o
FAILED: nvme.p/plugins_zns_zns.c.o
/usr/bin/gcc -Invme.p -I. -I.. -Iccan -I../ccan -I/usr/include/uuid -I/usr/include/json-c -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -O0 -fomit-frame-pointer -D_GNU_SOURCE -include config.h -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -MD -MQ nvme.p/plugins_zns_zns.c.o -MF nvme.p/plugins_zns_zns.c.o.d -o nvme.p/plugins_zns_zns.c.o -c ../plugins/zns/zns.c
../plugins/zns/zns.c: In function ‘zone_append’:
../plugins/zns/zns.c:1187:18: error: ‘struct nvme_zns_append_args’ has no member named ‘ilbrt_u64’
 1187 |                 .ilbrt_u64      = cfg.ref_tag,
      |                  ^~~~~~~~~
[31/47] Compiling C object nvme.p/nvme-print.c.o
FAILED: nvme.p/nvme-print.c.o
/usr/bin/gcc -Invme.p -I. -I.. -Iccan -I../ccan -I/usr/include/uuid -I/usr/include/json-c -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -O0 -fomit-frame-pointer -D_GNU_SOURCE -include config.h -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -MD -MQ nvme.p/nvme-print.c.o -MF nvme.p/nvme-print.c.o.d -o nvme.p/nvme-print.c.o -c ../nvme-print.c
../nvme-print.c: In function ‘nvme_show_registers_cc’:
../nvme-print.c:2628:17: warning: implicit declaration of function ‘NVME_CC_CRIME’; did you mean ‘NVME_CC_AMS’? [-Wimplicit-function-declaration]
 2628 |                 NVME_CC_CRIME(cc) ? "Enabled":"Disabled");
      |                 ^~~~~~~~~~~~~
      |                 NVME_CC_AMS
../nvme-print.c: In function ‘nvme_show_registers_crto’:
../nvme-print.c:2681:17: warning: implicit declaration of function ‘NVME_CRTO_CRIMT’ [-Wimplicit-function-declaration]
 2681 |                 NVME_CRTO_CRIMT(crto)/2 );
      |                 ^~~~~~~~~~~~~~~
../nvme-print.c:2683:17: warning: implicit declaration of function ‘NVME_CRTO_CRWMT’ [-Wimplicit-function-declaration]
 2683 |                 NVME_CRTO_CRWMT(crto)/2 );
      |                 ^~~~~~~~~~~~~~~
../nvme-print.c: In function ‘json_ctrl_registers’:
../nvme-print.c:2936:34: error: ‘NVME_REG_CRTO’ undeclared (first use in this function); did you mean ‘NVME_REG_CSTS’?
 2936 |         crto = mmio_read32(bar + NVME_REG_CRTO);
      |                                  ^~~~~~~~~~~~~
      |                                  NVME_REG_CSTS
../nvme-print.c:2936:34: note: each undeclared identifier is reported only once for each function it appears in
../nvme-print.c: In function ‘nvme_show_ctrl_registers’:
[... 94 lines cut ...]
../nvme.c:6744:28: error: ‘NVME_IO_STC’ undeclared (first use in this function); did you mean ‘NVME_IO_LR’?
 6744 |                 control |= NVME_IO_STC;
      |                            ^~~~~~~~~~~
      |                            NVME_IO_LR
../nvme.c:6766:49: error: ‘NVME_NVM_ELBAF_STS_MASK’ undeclared (first use in this function); did you mean ‘NVME_LBAF_RP_MASK’?
 6766 |                 sts = nvm_ns.elbaf[lba_index] & NVME_NVM_ELBAF_STS_MASK;
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~
      |                                                 NVME_LBAF_RP_MASK
[... cut ...]
 6782 |                 .reftag_u64     = cfg.ref_tag,
      |                  ^~~~~~~~~~
      |                  reftag
../nvme.c:6785:18: error: ‘struct nvme_io_args’ has no member named ‘sts’
 6785 |                 .sts            = sts,
      |                  ^~~
../nvme.c:6786:18: error: ‘struct nvme_io_args’ has no member named ‘pif’
 6786 |                 .pif            = pif,
      |                  ^~~
[46/47] Generating Documentation/nvme_zns_open_zone_man with a custom command
ninja: build stopped: subcommand failed.
```

igaw, 2022-08-01:
This looks like you haven't updated libnvme:

https://github.com/linux-nvme/nvme-cli#meson-tips

kloczek, 2022-08-01:
Here is the fix:
```patch
--- a/meson.build
+++ b/meson.build
@@ -44,7 +44,7 @@
 conf.set('SYSCONFDIR', '"@0@"'.format(sysconfdir))

 # Check for libnvme availability
-libnvme_dep = dependency('libnvme', required: true,
+libnvme_dep = dependency('libnvme', version: '>= 1.1', required: true,
                          fallback : ['libnvme', 'libnvme_dep'])

 # Check for libuuid availability
```
I think that it would be good to make new 2.1.1 release to avoid more reports like that.

igaw, 2022-08-01:
Yep, but that's not the only thing which is missing. we need also to define a weak `nvme_init_copy_range_f1` symbol to drop the hard linker dependency on the `LIBNVME_1_1` section.

closed 2022-08-01 by igaw

kloczek, 2022-08-01:
OK thank you 👍 
So my understanding is that soon will be 2.1.1 ? 😋

igaw, 2022-08-01:
Yeah, just fixing the release script to be able to produce a v2.1.1... brown bag release...

igaw, 2022-08-11:
And while at it we might need another release to fixup my brown bag release...

PR #1629 (2022-08-01): Fix release fallouts
Unfortunately, we have a couple of build fallouts.

Fixes #1628
PR #1636 (2022-08-11): wrapper: Call library version of nvme_init_copy_range_f1
The wrapper needs to call the library version of the function.

While at it, introduce preprocessor helpers to reduce the amount
boilerplate code.

Fixes: 7ccb7d29afa0 ("wrapper: Add weak nvme_init_copy_range_f1 symbol")

Fixes: #1628