linux-nvme/nvme-cli #1631: Unwanted side-effects with nvme discovery

martin-gpy, 2022-08-10:
Observing unwanted side-effects currently with nvme discover. For e.g., running a simple 'nvme discover' command ends up invoking connect-all as well, creating namespaces on the host:

```
# nvme list
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
```

```
# nvme discover -t fc -w nn-0x20000024ff19ab71:pn-0x21000024ff19ab71 -a nn-0x203400a098fd4ca6:pn-0x203800a098fd4ca6

Discovery Log Number of Records 6, Generation counter 34
=====Discovery Log Entry 0======
trtype:  fc
adrfam:  fibre-channel
subtype: current discovery subsystem
treq:    not specified
portid:  0
trsvcid: none
subnqn:  nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:discovery
traddr:  nn-0x203400a098fd4ca6:pn-0x203700a098fd4ca6
eflags:  explicit discovery connections, duplicate discovery information
=====Discovery Log Entry 1======
trtype:  fc
adrfam:  fibre-channel
subtype: current discovery subsystem
treq:    not specified
portid:  1
trsvcid: none
subnqn:  nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:discovery
traddr:  nn-0x203400a098fd4ca6:pn-0x203600a098fd4ca6
eflags:  explicit discovery connections, duplicate discovery information
=====Discovery Log Entry 2======
trtype:  fc
adrfam:  fibre-channel
subtype: current discovery subsystem
treq:    not specified
portid:  2
trsvcid: none
subnqn:  nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:discovery
traddr:  nn-0x203400a098fd4ca6:pn-0x203800a098fd4ca6
eflags:  explicit discovery connections, duplicate discovery information
=====Discovery Log Entry 3======
trtype:  fc
adrfam:  fibre-channel
subtype: nvme subsystem
treq:    not specified
portid:  0
trsvcid: none
subnqn:  nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:subsystem.subsys_119
traddr:  nn-0x203400a098fd4ca6:pn-0x203700a098fd4ca6
eflags:  not specified
=====Discovery Log Entry 4======
trtype:  fc
adrfam:  fibre-channel
subtype: nvme subsystem
treq:    not specified
portid:  1
trsvcid: none
subnqn:  nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:subsystem.subsys_119
traddr:  nn-0x203400a098fd4ca6:pn-0x203600a098fd4ca6
eflags:  not specified
=====Discovery Log Entry 5======
trtype:  fc
adrfam:  fibre-channel
subtype: nvme subsystem
treq:    not specified
portid:  2
trsvcid: none
subnqn:  nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:subsystem.subsys_119
traddr:  nn-0x203400a098fd4ca6:pn-0x203800a098fd4ca6
eflags:  not specified
```

```
# nvme list
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme1n1          /dev/ng1n1            816qoFSKNQMfAAAAAAAB NetApp ONTAP Controller                  1          10.74  GB /  10.74  GB      4 KiB +  0 B   FFFFFFFF
```

martin-gpy, 2022-08-10:
Seeing these entries in the messages file triggered for the above discover command:

```
...
kernel: [59153.128726][ T6226] nvme nvme0: NVME-FC{0}: create association : host wwpn 0x21000024ff19ab71  rport wwpn 0x203800a098fd4ca6: NQN "nqn.2014-08.org.nvmexpress.discovery"
kernel: [59154.048727][ T6226] nvme nvme0: queue_size 128 > ctrl maxcmd 32, reducing to maxcmd
kernel: [59154.048771][ T6226] nvme nvme0: NVME-FC{0}: controller connect complete
kernel: [59154.048816][ T6333] nvme nvme0: NVME-FC{0}: new ctrl: NQN "nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:discovery"
systemd[1]: Started NVMf auto-connect scan upon nvme discovery controller Events.
kernel: [59154.426283][ T6333] nvme nvme0: Removing ctrl: NQN "nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:discovery"
kernel: [59154.429587][ T6337] nvme_fc: nvme_fc_create_ctrl: nn-0x203400a098fd4ca6:pn-0x203700a098fd4ca6 - nn-0x20000024ff19ab71:pn-0x21000024ff19ab71 combination not found
kernel: [59154.433073][ T6039] nvme nvme1: NVME-FC{1}: create association : host wwpn 0x21000024ff19ab71  rport wwpn 0x203600a098fd4ca6: NQN "nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:subsystem.subsys_119"
kernel: [59156.009115][ T6039] nvme nvme1: NVME-FC{1}: controller connect complete
kernel: [59156.009169][ T6337] nvme nvme1: NVME-FC{1}: new ctrl: NQN "nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:subsystem.subsys_119"
kernel: [59156.013607][ T6065] nvme nvme2: NVME-FC{2}: create association : host wwpn 0x21000024ff19ab71  rport wwpn 0x203800a098fd4ca6: NQN "nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:subsystem.subsys_119"
kernel: [59157.527625][ T6065] nvme nvme2: NVME-FC{2}: controller connect complete
kernel: [59157.527681][ T6337] nvme nvme2: NVME-FC{2}: new ctrl: NQN "nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:subsystem.subsys_119"
systemd[1]: nvmf-connect@--device\x3dnvme0\t--transport\x3dfc\t--traddr\x3dnn-0x203400a098fd4ca6:pn-0x203800a098fd4ca6\t--trsvcid\x3dnone\t--host-traddr\x3dnn-0x20000024ff19ab71:pn-0x21000024ff19ab71.service: Deactivated successfully.
...
```

martin-gpy, 2022-08-10:
One can also observe spurious discovery controllers in the 'nvme list-subsys' output, despite having not created any persistent discovery controllers at all:

```
# nvme list-subsys
nvme-subsys1 - NQN=nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:subsystem.subsys_119
\
 +- nvme1 fc traddr=nn-0x203400a098fd4ca6:pn-0x203600a098fd4ca6,host_traddr=nn-0x20000024ff19ab71:pn-0x21000024ff19ab71 live
 +- nvme2 fc traddr=nn-0x203400a098fd4ca6:pn-0x203800a098fd4ca6,host_traddr=nn-0x20000024ff19ab71:pn-0x21000024ff19ab71 live
 +- nvme3 fc traddr=nn-0x203400a098fd4ca6:pn-0x203700a098fd4ca6,host_traddr=nn-0x20000024ff19ab70:pn-0x21000024ff19ab70 live
nvme-subsys0 - NQN=nqn.1992-08.com.netapp:sn.e6f0479b886211eca69d00a098fd4ca6:discovery
\
 +- nvme0 fc traddr=nn-0x203400a098fd4ca6:pn-0x203700a098fd4ca6,host_traddr=nn-0x20000024ff19ab70:pn-0x21000024ff19ab70 live
```

martin-gpy, 2022-08-10:
All these side-effects seem to be triggered from this recent commit:

udev: re-read the discovery log page when a discovery controller reconnected - https://github.com/linux-nvme/nvme-cli/commit/f86faaaa2a1ff319bde188dc8988be1ec054d238

igaw, 2022-08-11:
@sagigrimberg

sagi-lb, 2022-08-11:
Ugh... this needs to be reverted I guess. Need to come up with a better way to do this. Maybe if the discovery-controller is persistent only do this...

igaw, 2022-08-11:
Okay, let's revert it for the time being until we find another way to deal with your problem.

sagi-lb, 2022-08-11:
Maybe we need some more assistance from the driver to tell us about "reconnected" event and not just "connected"..
Or, it is up to the controller to send an AEN when the host reconnects to persistent discovery controllers...

igaw, 2022-08-11:
We had some discussion on the 'reconnect' event if I am not completely mistaken. There was some argument against 'reconnect'.

yizhanglinux, 2022-08-11:
yeah, this also impacts blktests nvme/030 failure,
https://lore.kernel.org/linux-block/CAHj4cs82ssuVX25yeHXhtqkkApxJbWDaoyOgq=0u5C4LWF2btg@mail.gmail.com/T/#u

sagi-lb, 2022-08-11:
Yes because it is stateful... I was primarily the source of the resistance...

I'm thinking about making this a controller-side fix, but its not exactly reasonable for the controller to track hosts that were connected at some point in the past to send a AEN... So I do think that this is something that the host should address.

PR #1635 (2022-08-11): Revert "udev: re-read the discovery log page when a discovery control…
…ler reconnected"

This reverts commit f86faaaa2a1ff319bde188dc8988be1ec054d238.

Observing unwanted side-effects with nvme discovery. For e.g., running
a simple 'nvme discover' command ends up invoking connect-all as well,
creating namespaces on the host.

One can also see spurious discovery controllers in the 'nvme
list-subsys' output, despite having not created any persistent
discovery controllers at all.

Reported-by: Martin George <Martin.George@netapp.com>

Fixes: #1631
closed 2022-08-11 by igaw