linux-nvme/nvme-cli #1640: Interpretting self-test-log

jaredmo, 2022-08-13:
Is there documentation on how to interpret the results in `self-test-log` ? See results below.

````
  Self Test Result[0]:
  Operation Result             : 0x7
  Self Test Code               : 2
  Segment Number               : 0x7
  Valid Diagnostic Information : 0x3
  Power on hours (POH)         : 0x6ec
  Namespace Identifier         : 0x1
  Failing LBA                  : 0
  Vendor Specific              : 0 0

igaw, 2022-08-16:
Check the specification chapter: "5.16.1.7 Device Self-test (Log Identifier 06h)"

https://nvmexpress.org/developers/nvme-specification/

There is a little bit of documentation what these values represent.

igaw, 2022-08-16:
BTW, you can also use the `-H` command line option to decode these values for you.

closed 2022-08-16 by igaw

imthenachoman, 2023-10-18:
@igaw Where do you use the `-H` option? I tried `nvme self-test-log /dev/nvme0 -H` and it says unrecognized option.

igaw, 2023-11-02:
Please do not hick jack old threads.

- My bad, the `-H` (aka `--human`) option is not supported (yet) by the `self-test-log` command.
- https://github.com/linux-nvme/nvme-cli/issues/1758#issuecomment-1373335611 has the pointers to the specification.

imthenachoman, 2023-11-03:
@igaw Got it. Thanks. And sorry. It's always confusing what the protocol is. Some repos want new issues, some want you to use old one. I wish there was a universally accepted best practice. I'll post a new issue.