linux-nvme/nvme-cli #1647: nvme-cli v2.x compile warning

martin-gpy, 2022-08-23:
Of late, I see the below compile warning when attempting to build the latest upstream nvme-cli v2.x (after successfully compiling & installing latest libnvme) on my host:

```
# ninja -C .build
ninja: Entering directory `.build'
[39/42] Compiling C object nvme.p/nvme.c.o
../nvme.c: In function ‘ns_rescan’:
../nvme.c:4389:19: warning: ‘dev’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  struct nvme_dev *dev;
                   ^~~
[42/42] Linking target nvme
```

This used to compile successfully with the earlier nvme-cli v2.x upstream builds. Any idea what broke now and how to get rid of this?

igaw, 2022-08-23:
What compiler version do you use? Any special configuration, e.g. debug build?

@jk-ozlabs

jk-ozlabs, 2022-08-23:
@martin-gpy I have a workaround for this (explicitly initialising `dev` to `NULL`), but I'd prefer to find the underlying cause here, as I'm not seeing any paths where `dev` can be used uninitialised. If I can replicate your setup here, that would definitely help.

Feel free to assign to me; I'll at least send a PR for the workaround if we can't find a specific cause soon.

jk-ozlabs, 2022-08-23:
@martin-gpy in the meantime, this is the workaround patch: https://github.com/linux-nvme/nvme-cli/commit/7f6362ae24e33ebf4dd9bcf5841d03385a0edca7

igaw, 2022-08-23:
IIRC, some distro compiler do enable additional checks/features to 'harden' the binaries. Maybe one of these options is getting the compiler confused.

 For example

  `CFLAGS=-Wmaybe-uninitialized meson .build --optimization=1`

results in 

```
[35/96] Compiling C object subprojects/libnvme/examples/mi-mctp.p/mi-mctp.c.o
../subprojects/libnvme/examples/mi-mctp.c: In function ‘do_action_endpoint’:
../subprojects/libnvme/examples/mi-mctp.c:665:16: warning: ‘rc’ may be used uninitialized [-Wmaybe-uninitialized]
  665 |         return rc;
      |                ^~
../subprojects/libnvme/examples/mi-mctp.c:637:13: note: ‘rc’ was declared here
  637 |         int rc;
      |             ^~
```

with my compiler: `gcc (SUSE Linux) 12.1.1 20220721`

The workaround is certainly an option if it is just these few places.

igaw, 2022-08-23:
At least we should document in the workaround commit which compiler triggers this warning.

martin-gpy, 2022-08-23:
> What compiler version do you use? Any special configuration, e.g. debug build?
> 

I'm running this on a SLES15 SP4 host. No special config used here. The gcc version is given below:

```
# gcc --version
gcc (SUSE Linux) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

martin-gpy, 2022-08-23:
> @martin-gpy in the meantime, this is the workaround patch: [7f6362a](https://github.com/linux-nvme/nvme-cli/commit/7f6362ae24e33ebf4dd9bcf5841d03385a0edca7)

I applied this patch, but it still threw a warning:

```
# ninja -C .build
ninja: Entering directory `.build'
[39/42] Compiling C object nvme.p/nvme.c.o
../nvme.c: In function ‘ns_rescan’:
../nvme.c:4389:19: warning: ‘dev’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  struct nvme_dev *dev;
                   ^~~
[42/42] Linking target nvme
```
This compiled successfully only after appending the above patch with the hunk described below:

 ```
  static int ns_rescan(int argc, char **argv, struct command *cmd, struct plugin *plugin)
  {
         const char *desc = "Rescans the NVMe namespaces\n";
 -       struct nvme_dev *dev;
 +       struct nvme_dev *dev = NULL;
         int err;
 
         OPT_ARGS(opts) = {
 ```
Now it compiled properly:

```
# ninja -C .build
ninja: Entering directory `.build'
[42/42] Linking target nvme
```

ikegami-t, 2022-08-23:
It seems that get_telemetry_log() also should be fixed as implemented as similar even if the waring not happend.

jk-ozlabs, 2022-08-24:
I'm still unable to reproduce here, even with gcc-12 + `-Wmaybe-uninitialized`, but I think I've discovered what may be triggering the warning:

If `parse_and_open` fails (ie., returns a negative value), we don't use `dev`:

```c
	err = parse_and_open(&dev, argc, argv, desc, opts);
	if (err < 0)
		goto ret;
```

However, positive return values would result in `dev` being used without initialisation, hence the warning.

A positive return value from `parse_and_open` is only possible if through the argument parsing failure path:

```c
	ret = argconfig_parse(argc, argv, desc, opts);
	if (ret)
		return ret;
```

(note the `if (ret)` rather than `if (ret < 0)`)

Now, `argconfig_parse()` should only fail with a negative errno value, so `ret` will always be negative on failure, but the compiler has no way to determine that, due to:

```c
	if (!long_opts || !short_opts) {
		fprintf(stderr, "failed to allocate memory for opts: %s\n",
				strerror(errno));
		ret = -errno;
		goto out;
	}
```
although it's convention for `errno` to only be positive (or zero), there's nothing that the compiler can rely on here.

So: it's impossible to hit this case (assuming `errno` is never negative), but we should still fix the warning.

Those `parse_and_open()` checks should also be failing on any non-zero return value, rather than just negative - this is a left-over from when `parse_and_open` used to return a file descriptor on success. I'll put together a patch.

jk-ozlabs, 2022-08-24:
OK, I've managed to replicate the warning using a `opensuse/leap:15` docker image, and have just pushed a PR for a fix. This addresses the warning there, but more testing would be great!

closed 2022-08-25 by igaw

igaw, 2022-08-25:
Thanks a lot! Really appreciated your work!

Ideally we find a way to run builds in a couple more containers. Last time I checked, github just offered ubuntu images and for anything else one needed to provide a self managed runner. It is a lot of work plus it generates real costs on our side. Maybe if we find some company to financially sponsors these builds I wouldn't mind to set it up. And on the plus side we could start using blktests for testing which would be a huge improvement.