linux-nvme/nvme-cli #1651: nvme0 failed to map. Did your kernel enable CONFIG_IO_STRICT_DEVMEM?

Kontr3X, 2022-08-25:
$ nvme show-regs /dev/nvme0n1
nvme0 failed to map. Did your kernel enable CONFIG_IO_STRICT_DEVMEM?

```
$ nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     48TS10BVTYST         KXG50ZNV512G TOSHIBA                     1         512.11  GB / 512.11  GB    512   B +  0 B   AAGA4106
/dev/nvme1n1     48TS10EFTYST         KXG50ZNV512G TOSHIBA                     1         512.11  GB / 512.11  GB    512   B +  0 B   AAGA4106
```

```
# uname -r
5.10.0-16-amd64
(Bullseye)
```

```
$ cat /sys/class/nvme/nvme0/device/resource
0x00000000ef100000 0x00000000ef103fff 0x0000000000140204
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
0x0000000000000000 0x0000000000000000 0x0000000000000000
```

`$ strace nvme show-regs nvme0
execve("/usr/sbin/nvme", ["nvme", "show-regs", "nvme0"], 0x7fff390096e0 /* 21 vars */) = 0
brk(NULL)                               = 0x562b42f8b000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=27342, ...}) = 0
mmap(NULL, 27342, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fcaed281000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libuuid.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0 #\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=30776, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fcaed27f000
mmap(NULL, 32792, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fcaed276000
mmap(0x7fcaed278000, 16384, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7fcaed278000
mmap(0x7fcaed27c000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7fcaed27c000
mmap(0x7fcaed27d000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7fcaed27d000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0@n\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1839792, ...}) = 0
mmap(NULL, 1852680, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fcaed0b1000
mprotect(0x7fcaed0d6000, 1662976, PROT_NONE) = 0
mmap(0x7fcaed0d6000, 1355776, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x25000) = 0x7fcaed0d6000
mmap(0x7fcaed221000, 303104, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x170000) = 0x7fcaed221000
mmap(0x7fcaed26c000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1ba000) = 0x7fcaed26c000
mmap(0x7fcaed272000, 13576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fcaed272000
close(3)                                = 0
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fcaed0ae000
arch_prctl(ARCH_SET_FS, 0x7fcaed0ae780) = 0
mprotect(0x7fcaed26c000, 12288, PROT_READ) = 0
mprotect(0x7fcaed27d000, 4096, PROT_READ) = 0
mprotect(0x562b42f85000, 8192, PROT_READ) = 0
mprotect(0x7fcaed2b2000, 4096, PROT_READ) = 0
munmap(0x7fcaed281000, 27342)           = 0
brk(NULL)                               = 0x562b42f8b000
brk(0x562b42fac000)                     = 0x562b42fac000
[... 267 lines cut ...]
write(2, "p", 1p)                        = 1
write(2, "u", 1u)                        = 1
write(2, "t", 1t)                        = 1
write(2, "_", 1_)                        = 1
write(2, "f", 1f)                        = 1
write(2, "o", 1o)                        = 1
write(2, "r", 1r)                        = 1
write(2, "m", 1m)                        = 1
write(2, "a", 1a)                        = 1
write(2, "t", 1t)                        = 1
write(2, " ", 1 )                        = 1
write(2, "=", 1=)                        = 1
write(2, "=", 1=)                        = 1
write(2, " ", 1 )                        = 1
write(2, "n", 1n)                        = 1
write(2, "o", 1o)                        = 1
write(2, "r", 1r)                        = 1
write(2, "m", 1m)                        = 1
write(2, "a", 1a)                        = 1
write(2, "l", 1l)                        = 1
write(2, "\n", 1
)                       = 1
exit_group(70)                          = ?
+++ exited with 70 +++
`

keithbusch, 2022-08-25:
We mmap resource0, not resource. The "resource" file just prints parameters and doesn't actually touch the memory it's describing.

The resource0 file is the one that provides the nvme controller register properties, and it's the only way we can access them on a pci device. I think the warning message is accurate: the kernel won't let user space map io memory.

Kontr3X, 2022-08-25:
I'm not sure how to interpret that statement. Maybe you can give me a hint where to start.

I try to use the nvme_exporter: https://github.com/yongseokoh/nvme_exporter/issues/4

keithbusch, 2022-08-25:
```
$ zgrep "CONFIG_STRICT_DEVMEM" /proc/config.gz
```

If the above says

```
CONFIG_STRICT_DEVMEM=y
```

Then your kernel won't let you run this command and there's nothing this tool can do about it. You would need to install a new kernel that doesn't have that config option enabled. Most distro kernels have that option enabled now-a-days, so you may need to manually compile one if you really need this command to work.

keithbusch, 2022-08-25:
Correction, I meant to grep for `CONFIG_IO_STRICT_DEVMEM`, which is different than `CONFIG_STRICT_DEVMEM`.

Kontr3X, 2022-08-25:
I don't have this file (/proc/config.gz) but it is enabled on the boot options (/boot/config-5.*).. I'm not sure, if we can just change these boot parameter then to "n". But it seems like you are right with it, it is enabled by default for security purpose probably. This is also good, at least in server environments that are online, but I will keep this in mind for debugging methods. Thank you very much so far! I'll close this issue with the note, that it's a feature, not a bug.

closed 2022-08-25 by Kontr3X