linux-nvme/nvme-cli #1658: segfault when using get-feature -f 0x0d

Mattwmaster58, 2022-08-31:
`sudo nvme get-feature /dev/nvme0 -f 0x0d -H` causes a segfault on my system. The segfault does not occur when the `-H` flag is omitted, ie, `sudo nvme get-feature /dev/nvme0 -f 0x0d` does not crash.
System details:

Kernel: 5.15.60-1-MANJARO 
CPU: AMD Ryzen 9 5900X (24) @ 3.700GHz 
Memory: 6154MiB / 64218MiB
SSD: WD_BLACK SN770 2TB (FW Revision 731030WD)

Maybe this can help as well:

```
 > sudo nvme get-feature /dev/nvme0 -f 0x0d -l 100                                                                                                                         
get-feature:0x0d (Host Memory Buffer), Current value:0x00000001
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: 00 20 00 00 00 50 df fe 00 00 00 00 08 00 00 00 ".....P.........."
0010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0060: 00 00 00 00                                      "....."
```

igaw, 2022-08-31:
What's the version of nvme-cli?

igaw, 2022-08-31:
Looks like `nvme_feature_show_fields()` is tripping over.

igaw, 2022-08-31:
bbc007d43d36 ("nvme: fix nvme get-feature with -H option")

Mattwmaster58, 2022-08-31:
> What's the version of nvme-cli?

2.0

Mattwmaster58, 2022-08-31:
> If this bbc007d43d36 ("nvme: fix nvme get-feature with -H option") is not fixing it, there is yet another case we have an invalid buffer access. Would it  possible to post the output from gdb bt? Thanks

I will look into this when I get the chance thanks for the quick reply!

Mattwmaster58, 2022-08-31:
Checked out `master` and issue appears to be resolved, thanks

closed 2022-08-31 by Mattwmaster58