linux-nvme/nvme-cli #1670: Make PDC the default option for targets that are TP 8013/8014 compliant

martin-gpy, 2022-09-07:
For targets that are already TP 8013/8014 compliant, the host should consider creating Persistent Discovery Controllers (PDC) as the default during NVMe discovery, instead of the regular short-lived non-PDC.

hreinecke, 2022-09-08:
... and this should be tied in with the detection of unique discovery subsystems (logging in with the default discovery subsystem NQN, figuring our the unique subsystem NQN, logout, and then re-login persistently with the unique subsystem NQN.)

martin-gpy, 2022-10-28:
A related ask - while making PDC the default option against a TP 8013/8014 compliant target, care should also be taken to ensure only a single PDC is created for all those records where DUPRETINFO is enabled in the discovery log page data i.e. one need not unnecessarily create PDCs for each record in the discovery log page data if DUPRETINFO is enabled.

igaw, 2023-01-20:
I've implemented this feature now (#1765). One thing which feels a bit odd is that 'unique discovery controller' means 'persistent discovery controller'. I haven't really figured out how else to say 'make this a  persistent discovery controller'.

PR #1765 (2023-01-24): Make unique discovery controllers persistent as default
This implemention changes the default behavior considerable:

- Try to use unique discovery controllers per default.
- Avoid connecting to already connected or 'duplicates' controllers during discovery
- Make unique discovery controllers persistent per default

The first two are backwards compatible, that means if the user isn't really using '--persistent' as command line option, the outcome should be the same. The last change is a bit daunting IMO. Is this really what the TP8013 mandates?

Fixes #1670
closed 2023-01-24 by igaw