linux-nvme/nvme-cli #1671: The namespace is getting attached automatically from the second path.

sarabjotsoit, 2022-09-07:
Steps:
# Create namespace with nvme cli 2.1.2.
# Attach the namespace from one controller.

Observation:
The namespace is getting attached automatically from the second path. The issue is not seen with the nvme-cli 1.13

sarabjotsoit, 2022-09-07:
Is this a new design?

igaw, 2022-09-08:
I don't think it's is purpose. Can you provide the exact commands you are issuing and some logs?

sarabjotsoit, 2022-09-09:
Hi, @igaw 

I believe it "lists 2 controllers though ns is attached from only one controller."

1. Connect the drives from 2 paths(controllers).
2. Create a namespace on drive and attach it from one path.
3. check "nvme list -v".
 
Actual result:
 
It lists 2 controllers though ns is attached from only one controller.
 
root@dcs:~# nvme list -v
Subsystem        Subsystem-NQN                                                                                    Controllers
---------------- ------------------------------------------------------------------------------------------------ ----------------
nvme-subsys9     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.9                      nvme9, nvme33
nvme-subsys8     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.8                      nvme8, nvme32
nvme-subsys7     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.7                      nvme7, nvme31
nvme-subsys6     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.6                      nvme6, nvme30
nvme-subsys5     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.5                      nvme5, nvme29
nvme-subsys4     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.4                      nvme4, nvme28
nvme-subsys3     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.3                      nvme3, nvme27
nvme-subsys24    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.24                     nvme48, nvme24
nvme-subsys23    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.23                     nvme47, nvme23
nvme-subsys22    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.22                     nvme46, nvme22
nvme-subsys21    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.21                     nvme45, nvme21
nvme-subsys20    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.20                     nvme44, nvme20
nvme-subsys2     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.2                      nvme26, nvme2
nvme-subsys19    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.19                     nvme43, nvme19
nvme-subsys18    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.18                     nvme42, nvme18
nvme-subsys17    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.17                     nvme41, nvme17
nvme-subsys16    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.16                     nvme40, nvme16
nvme-subsys15    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.15                     nvme39, nvme15
nvme-subsys14    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.14                     nvme38, nvme14
nvme-subsys13    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.13                     nvme37, nvme13
nvme-subsys12    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.12                     nvme36, nvme12
nvme-subsys11    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.11                     nvme35, nvme11
nvme-subsys10    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.10                     nvme34, nvme10
nvme-subsys1     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.1                      nvme25, nvme1
 
Device   SN                   MN                                       FR       TxPort Address        Subsystem    Namespaces
-------- -------------------- ---------------------------------------- -------- ------ -------------- ------------ ----------------
nvme9    A05D3C6B             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys9 nvme9n1
nvme33   A05D3C6B             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys9
nvme8    A06A4A6D             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys8
nvme32   A06A4A6D             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys8
nvme7    A05D3BC8             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys7 nvme7n1
nvme31   A05D3BC8             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys7
nvme6    A06A4A73             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys6 nvme6n1
nvme30   A06A4A73             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys6
nvme5    A05D3C22             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys5 nvme5n1
nvme29   A05D3C22             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys5
nvme4    A05D3CF5             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys4 nvme4n1
[... cut ...]
/dev/nvme14n1 /dev/ng14n1  1         15.36  TB /  15.36  TB      4 KiB +  0 B   nvme38, nvme14
/dev/nvme13n1 /dev/ng13n1  1          7.68  TB /   7.68  TB      4 KiB +  0 B   nvme37, nvme13
/dev/nvme12n1 /dev/ng12n1  1         15.36  TB /  15.36  TB      4 KiB +  0 B   nvme36, nvme12
/dev/nvme11n1 /dev/ng11n1  1          7.68  TB /   7.68  TB      4 KiB +  0 B   nvme35, nvme11
/dev/nvme10n1 /dev/ng10n1  1         15.36  TB /  15.36  TB      4 KiB +  0 B   nvme34, nvme10
/dev/nvme1n1 /dev/ng1n1   1          7.68  TB /   7.68  TB      4 KiB +  0 B   nvme25, nvme1

igaw, 2022-09-09:
I've should have used the word 'verbatim' instead of 'exact'. Trying to guess what you have typed on the shell is introducing another layer of possible misunderstandings. 

> * Connect the drives from 2 paths(controllers).
> 
> * Create a namespace on drive and attach it from one path.

Do I understand this correctly, the second step on the storage array. No nvme-cli command on the host?

> * check "nvme list -v".

Are there any kernel logs?

igaw, 2022-09-30:
No feedback. Closing.

closed 2022-09-30 by igaw