linux-nvme/nvme-cli #1672: FW Rev Field from nvme list not Being Updated

jeffreyalien, 2022-09-08:
The FW Rev field from the "nvme list" command is not being updated after a fw activate.  The fw revision field returned from the id-ctrl as well as the value returned by the fw-log command are both showing the new fw revision so I've verified the fw activate was successful.  I see with the new libnvme code, the fw revision is being read from the sysfs directory now.  Before the libnvme code it was read directly from the id-ctrl data.  

Is there an additional step required now to update/refresh the sysfs data?   If so, what would that be?  I've confirmed that a pci remove and rescan will update the sysfs data and subsequently the nvme list data but that doesn't see to be an acceptable long term solution.  

If there's no simpler way to update the sysfs data, can the libnvme code be changed back to read the fw rev from id-ctrl data?  Or is there other options?  I'm willing to make the fix, just not sure which way to go with it.

keithbusch, 2022-09-08:
This is a kernel bug. It should refresh the attribute automatically, but it is doing this only once. I'll add it to my to-do list.

jeffreyalien, 2022-09-08:
Thanks Keith.  I can verify the fix whenever it's available.

igaw, 2022-09-22:
Kernel bug, hence closing this bug report.

closed 2022-09-22 by igaw