linux-nvme/nvme-cli #1674: nvme-cli-monolithic: nvme connect-all broken against a TP 8013/8014 compliant target

martin-gpy, 2022-09-13:
With the latest nvme-cli-monolithic branch, nvme connect-all seems broken now when tested against a TP 8013/8014 compliant target as shown below:

```
# nvme connect-all -t fc -w nn-0x200000109b579d5e:pn-0x100000109b579d5e -a nn-0x202400a098fd4ca6:pn-0x202500a098fd4ca6
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
...
```

This seems to go in an indefinite loop.

martin-gpy, 2022-09-13:
Problem seems to stem from the following line in ctrl_matches_connectargs():fabrics.c:

```
...
        cargs.subsysnqn = nvme_get_ctrl_attr(path, "subsysnqn");
        cargs.transport = nvme_get_ctrl_attr(path, "transport");
        cargs.subsystype = nvme_get_ctrl_attr(path, "subsystype"); <------
        if (!cargs.subsystype) {
                if (!strcmp(cargs.subsysnqn, NVME_DISC_SUBSYS_NAME))
                        cargs.subsystype = strdup("discovery");
                else
                        cargs.subsystype = strdup("nvm");
        }
...
```
nvme_get_ctrl_attr(path, "subsystype") always returns NULL since there is no 'subsystype' attribute defined at /sys/class/nvme/nvmeX/.  Instead of searching the 'subsystype' attribute at /sys/class/nvme/nvmeX/, it should have accessed this attribute at /sys/class/nvme-subsystem/nvme-subsysX/subsystype.

igaw, 2022-09-20:
```shell
# ./nvme connect-all --transport=fc --traddr=nn-0x201700a09890f5bf:pn-0x201900a09890f5bf --host-traddr=nn-0x200000109b579ef3:pn-0x100000109b579ef3
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory
Failed to write to /dev/nvme-fabrics: No such file or directory

# nvme list
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme3n1          /dev/ng3n1            80-AABMxwn3xAAAAAAAF NetApp ONTAP Controller                  1          34.36  GB /  34.36  GB      4 KiB +  0 B   FFFFFFFF
/dev/nvme1n2          /dev/ng1n2            80-AABMxwn3xAAAAAAAB NetApp ONTAP Controller                  2          34.36  GB /  34.36  GB      4 KiB +  0 B   FFFFFFFF
/dev/nvme1n1          /dev/ng1n1            80-AABMxwn3xAAAAAAAB NetApp ONTAP Controller                  1          34.36  GB /  34.36  GB      4 KiB +  0 B   FFFFFFFF

# nvme list -v
Subsystem        Subsystem-NQN                                                                                    Controllers
---------------- ------------------------------------------------------------------------------------------------ ----------------
nvme-subsys3     nqn.1992-08.com.netapp:sn.d646dc63336511e995cb00a0988fb732:subsystem.nvme-svm-dolin-ana_subsystem_mwilck nvme4, nvme3
nvme-subsys1     nqn.1992-08.com.netapp:sn.d646dc63336511e995cb00a0988fb732:subsystem.nvme-svm-dolin-ana_subsystem nvme2, nvme1
nvme-subsys0     nqn.1992-08.com.netapp:sn.d646dc63336511e995cb00a0988fb732:discovery                             nvme0

Device   SN                   MN                                       FR       TxPort Address        Subsystem    Namespaces      
-------- -------------------- ---------------------------------------- -------- ------ -------------- ------------ ----------------
nvme4    80-AABMxwn3xAAAAAAAF NetApp ONTAP Controller                  FFFFFFFF fc     traddr=nn-0x201700a09890f5bf:pn-0x201900a09890f5bf,host_traddr=nn-0x200000109b579ef3:pn-0x100000109b579ef3 nvme-subsys3 nvme3n1
nvme3    80-AABMxwn3xAAAAAAAF NetApp ONTAP Controller                  FFFFFFFF fc     traddr=nn-0x201700a09890f5bf:pn-0x201800a09890f5bf,host_traddr=nn-0x200000109b579ef3:pn-0x100000109b579ef3 nvme-subsys3 nvme3n1
nvme2    80-AABMxwn3xAAAAAAAB NetApp ONTAP Controller                  FFFFFFFF fc     traddr=nn-0x201700a09890f5bf:pn-0x201900a09890f5bf,host_traddr=nn-0x200000109b579ef3:pn-0x100000109b579ef3 nvme-subsys1 nvme1n2, nvme1n1
nvme1    80-AABMxwn3xAAAAAAAB NetApp ONTAP Controller                  FFFFFFFF fc     traddr=nn-0x201700a09890f5bf:pn-0x201800a09890f5bf,host_traddr=nn-0x200000109b579ef3:pn-0x100000109b579ef3 nvme-subsys1 nvme1n2, nvme1n1
nvme0    1158458              NetApp ONTAP Controller                  FFFFFFFF fc     traddr=nn-0x201700a09890f5bf:pn-0x201800a09890f5bf,host_traddr=nn-0x200000109b579ef3:pn-0x100000109b579ef3 nvme-subsys0 

Device       Generic      NSID     Usage                      Format           Controllers     
------------ ------------ -------- -------------------------- ---------------- ----------------
/dev/nvme3n1 /dev/ng3n1   1         34.36  GB /  34.36  GB      4 KiB +  0 B   nvme4, nvme3
/dev/nvme1n2 /dev/ng1n2   2         34.36  GB /  34.36  GB      4 KiB +  0 B   nvme2, nvme1
/dev/nvme1n1 /dev/ng1n1   1         34.36  GB /  34.36  GB      4 KiB +  0 B   nvme2, nvme1
```

igaw, 2022-10-18:
After offline discussion with @martin-gpy and @hreinecke, we decided to cease the activities on this topic. Too big effort to get everything sorted out.

closed 2022-10-18 by igaw