linux-nvme/nvme-cli #1680: Failed to build subproject libnvme

Dicridon, 2022-09-21:
Hi,

I'm trying to build the latest nvme-cli on Debian 9, but I have the following errors when building subproject `libnvme`
```
Reason: subprojects/libnvme/libnvme/meson.build:29: Include directory to be added is not an include directory object.
```
The content of `subprojects/libnvme/libnvme/meson.build:29` is as follows:
[image]
It seems the errors are trigger by wrong arguments to `include_directories`, but I'm not sure if this is caused by wrong configuration by me or the syntax change of meson.

Currently I'm using `meson 0.49.2`, can you give me any tips to debug this issue?

igaw, 2022-09-21:
The documentation says extension_module has the same arguments as https://mesonbuild.com/Reference-manual_functions.html#shared_module:

include_directories          : [list](https://mesonbuild.com/Reference-manual_elementary_list.html)[[inc](https://mesonbuild.com/Reference-manual_returned_inc.html) | [str](https://mesonbuild.com/Reference-manual_elementary_str.html)]                                                         # one or more objects created with the [include_directories()](https://mesonbuild.com/Reference-manual_functions.html#include_directories) function,

Maybe try to extend incdir with the internal_incdir and only pass one object to the extension_module?

Dicridon, 2022-09-21:
Thank you for your timely reply. I tried your tip, but it didn't work. The same error occurred. If I replace the list `[incdir, internal_incdir]` with `'a_directory_in_string'`, the error disappears (and of course the compilation fails too).

I see a similar issue in another repo and the conclusion is that only strings can be used, which is contradictory to the meson reference. So I guess maybe this is some meson version issue (maybe in some old version only strings are supported) but I'm not sure which version I should use.

Dicridon, 2022-09-21:
Or maybe we don't have to compile `nvme-cli` from source. All I want is the ZNS related subcommands, which is supported since `nvme-cli 1.12` as stated here https://zonedstorage.io/docs/tools/zns. I installed the `nvme 1.12` using a deb package on my Debian 10 machine (this is the latest package I can find for Debian 10), but the `nvme zns` command still reported `ERROR: Invalid sub-command zns`. Are there any other kernel modules or packages required for zns subcommand?

igaw, 2022-09-21:
One thing you could try is to build libnvme independent and not via the subproject method.