linux-nvme/nvme-cli #1681: Set feature for 0xD Host Memory Buffer, CLI expects buffer

1pandaman, 2022-09-21:
According to the NVMe spec FID 0xD uses a databuffer for get feature but not set feature, but a set feature on the cli will wait on user input if a data buffer is not provided.

igaw, 2022-09-22:
Can you post the commend you are using?

1pandaman, 2022-09-22:
sudo nvme set-feature /dev/nvme0 --feature-id=0xD --value=0

keithbusch, 2022-09-22:
Yeah, it needs a buffer, so if you don't provide a file or input stream, then you get to write it from stdin. What do you want to happen instead?

This feature though, and every feature that's owned by the kernel driver, seems like a bad idea to modify from user space unless part of some kind of disruptive test.

1pandaman, 2022-09-22:
I was expecting not to need a buffer as the nvme spec states it is doesn't require one. But if the cli requires it is sending a 0 filled buffer effectively equivalent to not sending a buffer?

keithbusch, 2022-09-22:
Oh, right, the set side for this feature doesn't require a buffer in the traditional sense. It's just the "get" side of this feature ID that needs a host buffer. I suppose we should disable the buffer check in that case, though if you're turning the feature on, it does take a descriptor list. That is a type of host buffer, but not the kind that user space can readily supply, so it's not really a safe feature to be messing with.

igaw, 2022-10-27:
The fix is in so closing the issue. Thanks!

closed 2022-10-27 by igaw