linux-nvme/nvme-cli #1687: workflow: Add addtitional build targets

igaw, 2022-09-26:
Add additional build targets such a cross build and static builds, likewise we have in [libnvme](https://github.com/linux-nvme/libnvme/pull/479)

jk-ozlabs, 2022-09-26:
I'm working on a change for this, but hitting an issue with the libnvme dependency:

 - we need to fallback to the wrapped libnvme
 - that wrapped libnvme needs `-Dpython=false` for the `meson setup`

without `-Dpython=false`, we get:

```
/usr/bin/powerpc64le-linux-gnu-gcc -Isubprojects/libnvme/libnvme/c2308c5@@_nvme.cpython-38-x86_64-linux-gnu@sha -Isubprojects/libnvme/libnvme -I../subprojects/libnvme/libnvme -Isubprojects/libnvme -I../subprojects/libnvme -Isubprojects/libnvme/ccan -I../subprojects/libnvme/ccan -Isubprojects/libnvme/src -I../subprojects/libnvme/src -Isubprojects/libnvme/internal -I../subprojects/libnvme/internal -I/usr/include/uuid -I/usr/include/json-c -I/usr/include/python3.8 -I/usr/include/x86_64-linux-gnu/python3.8 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Werror -std=gnu99 -O3 -fomit-frame-pointer -D_GNU_SOURCE -include internal/config.h -fPIC -MD -MQ 'subprojects/libnvme/libnvme/c2308c5@@_nvme.cpython-38-x86_64-linux-gnu@sha/meson-generated_.._nvme_wrap.c.o' -MF 'subprojects/libnvme/libnvme/c2308c5@@_nvme.cpython-38-x86_64-linux-gnu@sha/meson-generated_.._nvme_wrap.c.o.d' -o 'subprojects/libnvme/libnvme/c2308c5@@_nvme.cpython-38-x86_64-linux-gnu@sha/meson-generated_.._nvme_wrap.c.o' -c subprojects/libnvme/libnvme/nvme_wrap.c
In file included from /usr/include/python3.8/Python.h:8,
                 from subprojects/libnvme/libnvme/nvme_wrap.c:154:
/usr/include/python3.8/pyconfig.h:68:14: fatal error: powerpc64le-linux-gnu/python3.8/pyconfig.h: No such file or directory
   68 | #    include <powerpc64le-linux-gnu/python3.8/pyconfig.h>
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

(even if we add a `setup-python@v4` action, that doesn't seem suitable for cross compilation)

igaw, 2022-09-26:
Does this also happen with an additional `-Dlibnvme:python=false` ?

jk-ozlabs, 2022-09-26:
ooh, didn't know about that syntax, I'll check.

igaw, 2022-09-26:
Fancy feature of meson :) You can pass down config flags to subproject with the `-Dprefix:`

jk-ozlabs, 2022-09-26:
Yep, that looks good, thanks! This seems to pop a bug on powerpc64, so I'll include a fix in the PR too.