linux-nvme/nvme-cli #1689: Default values for trsvcid for TCP transport not working

igaw, 2022-09-26:
```shell
# ./nvme discover -t tcp -a lithium 
failed to add controller, error Unknown error -1

# ./nvme discover -t tcp -a lithium -s 4420

Discovery Log Number of Records 2, Generation counter 3
=====Discovery Log Entry 0======
[...]
```

martin-gpy, 2022-09-28:
What's the exact issue here? 

Is it that the libnvme/nvme-cli doesn't pass the default trsvcid of 8009 for NVMe/TCP discovery? Or that the target fails the NVMe/TCP discovery controller creation if using trsvcid of 8009?

igaw, 2022-09-28:
I don't know yet. I haven't had the time to look into it, just opened the issue for tracking. The target is Linux's target implementation (nvmet).

igaw, 2022-09-29:
My configuration of the soft target was wrong. There is no discovery controller on port 8009, hence it fails.

closed 2022-09-29 by igaw