linux-nvme/nvme-cli #1695: how to get NVMe disk transport PCIe 3/4 support status and current link status?

wangyugui, 2022-09-30:
Hi,

how to get NVMe disk transport PCIe 3/4 support status and current link status?

Best regards

keithbusch, 2022-09-30:
The lspci utility is better suited to get pcie link status than this program.

wangyugui, 2022-09-30:
could we add 'System Bus (PCI Express) Registers' output for 'nvme show-regs'?

igaw, 2022-09-30:
Keep in mind, that show-regs only work with kernel `CONFIG_IO_STRICT_DEVMEM=n` configs. As far I know, all popular distro have it enabled. So unless you ship your own kernel, it would not show.

igaw, 2022-09-30:
Do you mean the PCI registers? If so, no way.

wangyugui, 2022-09-30:
I test that 'nvme show-regs' works when the NVMe disk is connected through a PCIe switch (A card).

In the spec 'NVM-Express 1.4b'/ '2 System Bus (PCI Express) Registers',
'nvme show-regs' output the part 'Start 00h~ End 3Fh PCI Header'?

can we output the part of 'Start PXCAP~ End PXCAP+29h PCI Express Capability' if the part is enabled?

wangyugui, 2022-09-30:
there seems some support of PXCAP in
https://github.com/intel/cNVMe

keithbusch, 2022-09-30:
Let's say you want to get the link width and speed of `/dev/nvme0n1`. Just run:

```
$ cat /sys/block/nvme0n1/device/device/current_link_{speed,width}
8.0 GT/s PCIe
4
```

keithbusch, 2022-09-30:
> But NVMe disk maybe connected to PCIe switch. In that case, lspci fail to get the status of NVMe disk.

Why on earth does a switch causing lspci to fail? It should work just fine, and I have many setups with switches that don't have a problem with that. The `nvme list -v` command already reports the PCIe B:D.f, so just feed that into `lspci -vv -s <B:D.f>`, and you have exactly what you want.

I would really prefer to avoid duplicating functionality of transport specific protocol tools as much as possible. If it's outside the nvme spec, I don't want this tool taking responsibility for it. The closes thing you're looking for in NVMe specs comes from NVMe-MI, but I have yet to find a device that supports that command set out-of-band.

wangyugui, 2022-10-01:
sorry.

I though KIOXIA CM5 is PCIe4 SSD, but in fact it is a PCIe3 SSD.

I test CM6(PCIe4) SSD, both lspci and 'cat /sys/block/nvme0n1/device/device/current_link_{speed,width}' works as expected.

'max_link_speed 16.0 GT/s' is confirmed, even when it is connected to PCIe3 switch PEX8734.

Thanks a lot.

closed 2022-10-01 by wangyugui