linux-nvme/nvme-cli #1700: what does 0x2002 mean in status_field?

FirstLoveLife, 2022-10-16:
Hi, my fs(btrfs) recently oftern become read-only when playing like 0ad, I doubt my nvme has some issues, so I check it with nvme-cli:
```
firstlove@chenli ~> sudo smartctl -A /dev/nvme0n1p3
smartctl 7.3 2022-02-28 r5338 [x86_64-linux-6.0.1-arch2-1] (local build)
Copyright (C) 2002-22, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF SMART DATA SECTION ===
SMART/Health Information (NVMe Log 0x02)
Critical Warning:                   0x00
Temperature:                        49 Celsius
Available Spare:                    100%
Available Spare Threshold:          5%
Percentage Used:                    18%
Data Units Read:                    46,120,988 [23.6 TB]
Data Units Written:                 343,649,247 [175 TB]
Host Read Commands:                 1,184,731,427
Host Write Commands:                3,228,585,654
Controller Busy Time:               4,397
Power Cycles:                       922
Power On Hours:                     11,374
Unsafe Shutdowns:                   292
Media and Data Integrity Errors:    0
Error Information Log Entries:      1,620
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0

firstlove@chenli ~> sudo nvme error-log --log-entries=1 /dev/nvme0n1p3
Error Log Entries for device:nvme0n1p3 entries:1
.................
 Entry[ 0]   
.................
error_count     : 1620
sqid            : 0
cmdid           : 0x10
status_field    : 0x2002(Invalid Field in Command: A reserved coded value or an unsupported value in a defined field)
phase_tag       : 0
parm_err_loc    : 0x28
lba             : 0
nsid            : 0
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
cs              : 0
trtype_spec_info: 0
.................
```

I cannot tell from the log whether my nvme is healthy, what does 0x2002 mean in status_field?

keithbusch, 2022-10-16:
The high order 2 is the "more", and that says the controller created a log entry for the error. The status is really just the lower order bits, 0x002, which is invalid field error.

FirstLoveLife, 2022-10-16:
Hi @keithbusch 

does "invalid field error" mean there are some features this NVME doesn't implement?
is this error critical?

keithbusch, 2022-10-16:
It's just saying the controller doesn't support some feature, but I can't tell what from the log. It's an admin command so most likely some harmless optional command mode.

closed 2022-10-18 by igaw