linux-nvme/nvme-cli #1701: meson: future deprecated feature warning

igaw, 2022-10-18:
```
NOTICE: Future-deprecated features used:
 * 0.56.0: {'meson.build_root'}

```

eli-schwartz, 2022-10-19:
This is defined at https://github.com/linux-nvme/nvme-cli/blob/53ff88037e6c95726bfe283bca582e7107485094/tests/meson.build#L46

The Meson warning is because build_root() is considered "ugly", it's not obvious that it really means global_build_root() and has surprising behavior in subprojects. Most people probably want project_build_root().

In your case, I think you want to replace `meson.build_root() + '/tests'` with `meson.current_build_dir()` and avoid string joining *and* deprecated functions, by using a function that has *always* been correct to use.

igaw, 2022-10-20:
Thanks for the feedback. IIRC, I avoided `meson.current_build_dir()` because I though it bumps our meson version requirement. But it doesn't seem the case.

igaw, 2022-10-20:
Ah I remember why I opted for `meson.build_root()`. 

https://github.com/linux-nvme/nvme-cli/blob/53ff88037e6c95726bfe283bca582e7107485094/tests/meson.build#L47

would like to use `meson.project_build_root()` but it's not available before 0.56.0. 

Let me figure out how I could handle this correctly.

igaw, 2022-10-20:
I suppose this should be okay?

```meson
if meson.version().version_compare('>= 0.56')
    nvmecli_path = meson.project_build_root()
else
    nvmecli_path = meson.build_root()
endif
```

igaw, 2022-10-20:
Hmm, I think we might want to move those tests which rely on the nvme-cli binary to different directory/project anyway. These are tests for HW vendors. As they are all depending to be executed as root and will overwrite data on /dev/nvme0.

eli-schwartz, 2022-10-20:
> I suppose this should be okay?

That should be fine, but you can also freely choose to ignore the notice. It's specifically not logged as a warning, until your minimum version matches the version its replacement was added in.

igaw, 2022-10-21:
Right, I think I'll add it as I try to avoid bumping the minimum version requirement.  It might be silly but I got burned in the past with updating the dependency too fast in other projects. At least for the 1.x version I'd like stay at this. For the upcoming 2.x version I am fine with updating it though.

PR #1705 (2022-10-21): build: Use meson.current_build_dir() for tests
Meson reports future deprecation of meson.build_root().  The recommended replacement is to use meson.current_build_dir().

Fixes: #1701
closed 2022-10-21 by igaw