linux-nvme/nvme-cli #1717: 2.2: build fails because missing libuuid checking dependencies

kloczek, 2022-10-30:
Looks like it is missing checking libuuid detpendencies
```console
+ /usr/bin/meson --buildtype=plain --prefix=/usr --libdir=/usr/lib64 --libexecdir=/usr/libexec --bindir=/usr/bin --sbindir=/usr/sbin --includedir=/usr/include --datadir=/usr/share --mandir=/usr/share/man --infodir=/usr/share/info --localedir=/usr/share/locale --sysconfdir=/etc --localstatedir=/var --sharedstatedir=/var/lib --wrap-mode=nodownload --auto-features=enabled . x86_64-redhat-linux-gnu -D docs=man -D docs-build=true -D htmldir=/usr/share/doc/nvme-cli -D systemddir=/usr/lib/systemd/system -D udevrulesdir=/usr/lib/udev/rules.d
The Meson build system
Version: 0.63.3
Source dir: /home/tkloczko/rpmbuild/BUILD/nvme-cli-2.2
Build dir: /home/tkloczko/rpmbuild/BUILD/nvme-cli-2.2/x86_64-redhat-linux-gnu
Build type: native build
Project name: nvme-cli
Project version: 2.2
C compiler for the host machine: /usr/bin/gcc (gcc 12.2.1 "gcc (GCC) 12.2.1 20220819 (Red Hat 12.2.1-2)")
C linker for the host machine: /usr/bin/gcc ld.bfd 2.39-3
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: /usr/bin/pkg-config (1.9.3)
Run-time dependency libnvme found: YES 1.1
Run-time dependency libnvme-mi found: YES 1.1
Run-time dependency json-c found: YES 0.16
Run-time dependency zlib found: YES 1.2.11.zlib-ng
Has header "hugetlbfs.h" : NO
Checking if "__builtin_type_compatible_p" compiles: YES
Checking if "typeof" compiles: YES
Checking if "byteswap.h" compiles: YES
Checking if "bswap64" : links: YES
Checking if "isblank" : links: YES
Checking if "sys/random.h" compiles: YES
Configuring config.h using configuration
Configuring nvme.spec using configuration
Configuring discovery.conf using configuration
Configuring 70-nvmf-autoconnect.conf using configuration
Configuring nvmefc-boot-connections.service using configuration
Configuring nvmf-autoconnect.service using configuration
Configuring nvmf-connect.target using configuration
Configuring nvmf-connect@.service using configuration
Configuring 70-nvmf-autoconnect.rules using configuration
Configuring 71-nvmf-iopolicy-netapp.rules using configuration
Program nose2 found: NO
Program python3 found: YES (/usr/bin/python3)
Program mypy found: YES (/usr/bin/mypy)
Program flake8 found: YES (/usr/bin/flake8)
Program autopep8 found: NO
Program isort found: YES (/usr/bin/isort)
Message: autopep8 or isort not found. Python formating disabled
Program asciidoc found: YES (/usr/bin/asciidoc)
Program xmlto found: YES (/usr/bin/xmlto)
Configuring nvme.msubst using configuration
Configuring nvme-admin-passthru.msubst using configuration
Configuring nvme-ana-log.msubst using configuration
Configuring nvme-attach-ns.msubst using configuration
Configuring nvme-boot-part-log.msubst using configuration
Configuring nvme-capacity-mgmt.msubst using configuration
Configuring nvme-changed-ns-list-log.msubst using configuration
Configuring nvme-cmdset-ind-id-ns.msubst using configuration
Configuring nvme-compare.msubst using configuration
Configuring nvme-connect.msubst using configuration
Configuring nvme-connect-all.msubst using configuration
Configuring nvme-copy.msubst using configuration
Configuring nvme-create-ns.msubst using configuration
Configuring nvme-delete-ns.msubst using configuration
Configuring nvme-dera-stat.msubst using configuration
Configuring nvme-detach-ns.msubst using configuration
Configuring nvme-device-self-test.msubst using configuration
Configuring nvme-dim.msubst using configuration
Configuring nvme-dir-receive.msubst using configuration
Configuring nvme-dir-send.msubst using configuration
Configuring nvme-disconnect.msubst using configuration
Configuring nvme-disconnect-all.msubst using configuration
Configuring nvme-discover.msubst using configuration
Configuring nvme-dsm.msubst using configuration
Configuring nvme-effects-log.msubst using configuration
[... 197 lines cut ...]
                 from ../tests/test-uint128.c:7:
/usr/include/nvme/tree.h:18:10: fatal error: uuid.h: No such file or directory
   18 | #include <uuid.h>
      |          ^~~~~~~~
compilation terminated.
```
```console
[tkloczko@devel-g2v SPECS]$ rpm -qal | grep uuid.h
/usr/include/linux/uuid.h
/usr/src/debug/pipewire-0.3.51-2.fc35.x86_64/pipewire-jack/jack/uuid.h
/usr/src/debug/glib2-2.72.1-2.fc35.x86_64/glib/guuid.h
/usr/include/glib-2.0/glib/guuid.h
/usr/include/uuid/uuid.h
[tkloczko@devel-g2v SPECS]$ cat /usr/lib64/pkgconfig/uuid.pc
prefix=/usr
exec_prefix=/usr
libdir=/usr/lib64
includedir=/usr/include

Name: uuid
Description: Universally unique id library
Version: 2.38.1
Cflags: -I${includedir}/uuid
Libs: -L${libdir} -luuid
```

eli-schwartz, 2022-10-30:
In 4901f97b7df3c192cf397cade82e5fa96df55e95 the dependency on libuuid got removed. It corresponds to https://github.com/linux-nvme/libnvme/commit/71c25d1cf741babf65891f71a0d9669b7a2369a1 as well.

You're using the 2.2 tag of nvme-cli, but the 1.1 tag, not 1.2, of libnvme -- and apparently that old libnvme version had a runtime -devel dependency on libuuid, but did not expose that as a pkg-config dependency? Hmm.

The solution ultimately is to just update, as I doubt the project is going to release a 1.1.1 bugfix for this.

kloczek, 2022-10-30:
If it is bug in libnvme it shoud be proper dependency checking in nvme-cli.

igaw, 2022-11-02:
> You're using the 2.2 tag of nvme-cli, but the 1.1 tag, not 1.2, of libnvme -- and apparently that old libnvme version had a runtime -devel dependency on libuuid, but did not expose that as a pkg-config dependency? Hmm.

It seems we are missing all the dependency in the spec file.

> The solution ultimately is to just update, as I doubt the project is going to release a 1.1.1 bugfix for this.

Not sure what's the point in release a bugfix libnvme. If libnvme needs to be touched, why not just upgrade it to 1.2?

igaw, 2022-11-02:
Note the pkg-config template in the root path is a left over which we just forgot to remove. When using the pkgconfig module from Meson, it generates a `Requires.private: json-c >= 0.13, openssl >= 1.1.0` entry. This looks reasonable but then I don't have a deep pkg-config know-how.

PR #1719 (2022-11-02): build: Update libnvme version dependency to 1.2
With the drop of libuuid we need also the corresponding libnvme library which provides its own uuid types.

Fixes: #1717
closed 2022-11-02 by igaw