linux-nvme/nvme-cli #1733: "nvme wdc id-ctrl" showing wrong data for "wdc vsn"

sarabjotsoit, 2022-11-17:
Steps:
1. Connect the drive and MI device to the host server.
2. Run "nvme wdc id-ctrl " for WD Drive and MI device(s).

Observation:
1. For the MI device "wdc vsn " showing as "Data24", but without the wdc plugin we are getting the proper product name as "Openflex Data 24".
2.  For the WD drive, the vsn value for the drive is showing NULL instead of the model number.

Logs::

root@dcs:~# nvme wdc id-ctrl /dev/nvme13n1
NVME Identify Controller:
vid       : 0x1c58
ssvid     : 0x1c58
sn        : USCOS02622QA0008BC
mn        : OpenFlex Data24
fr        : 1.0.2345
rab       : 1
ieee      : 000cca
cmic      : 0
mdts      : 5
cntlid    : 0x800e
ver       : 0x10300
rtd3r     : 0
rtd3e     : 0
oaes      : 0
ctratt    : 0x40
rrls      : 0
cntrltype : 0
fguid     :
crdt1     : 0
crdt2     : 0
crdt3     : 0
nvmsr     : 0
vwci      : 2
mec       : 0
oacs      : 0x4
acl       : 3
aerl      : 3
frmw      : 0x3
lpa       : 0x4
elpe      : 0
npss      : 0
avscc     : 0
apsta     : 0
wctemp    : 0
cctemp    : 0
mtfa      : 0
hmpre     : 0
hmmin     : 0
tnvmcap   : 16777216
unvmcap   : 0
rpmbs     : 0
edstt     : 0
dsto      : 0
fwug      : 255
kas       : 10
hctma     : 0
mntmt     : 0
mxtmt     : 0
sanicap   : 0
hmminds   : 0
hmmaxd    : 0
nsetidmax : 0
endgidmax : 0
anatt     : 0
anacap    : 0
anagrpmax : 0
nanagrpid : 0
pels      : 0
[... 544 lines cut ...]
ps      6 : mp:19.00W operational enlat:50 exlat:50 rrt:6 rrl:6
            rwt:6 rwl:6 idle_power:- active_power:-
            active_power_workload:-
ps      7 : mp:18.00W operational enlat:50 exlat:50 rrt:7 rrl:7
            rwt:7 rwl:7 idle_power:- active_power:-
            active_power_workload:-
ps      8 : mp:17.00W operational enlat:50 exlat:50 rrt:8 rrl:8
            rwt:8 rwl:8 idle_power:- active_power:-
            active_power_workload:-
ps      9 : mp:16.00W operational enlat:50 exlat:50 rrt:9 rrl:9
            rwt:9 rwl:9 idle_power:- active_power:-
            active_power_workload:-
ps     10 : mp:15.00W operational enlat:50 exlat:50 rrt:10 rrl:10
            rwt:10 rwl:10 idle_power:- active_power:-
            active_power_workload:-
ps     11 : mp:14.00W operational enlat:50 exlat:50 rrt:11 rrl:11
            rwt:11 rwl:11 idle_power:- active_power:-
            active_power_workload:-
ps     12 : mp:13.00W operational enlat:50 exlat:50 rrt:12 rrl:12
            rwt:12 rwl:12 idle_power:- active_power:-
            active_power_workload:-
ps     13 : mp:12.00W operational enlat:50 exlat:50 rrt:13 rrl:13
            rwt:13 rwl:13 idle_power:- active_power:-
            active_power_workload:-
wdc vsn : NULL

igaw, 2022-11-17:
@jeffreyalien could have a look at this report? Thanks!

jeffreyalien, 2022-11-17:
The wdc id-ctrl plugin command was added to display a vendor specific serial number that was supplied for a small number of nvme drives.  Most of the WD drives don't populate a VSN value and therefore will return NULL for that value.  I'd recommend using the base id-ctrl command vs the wdc plugin version.

igaw, 2022-11-18:
Thanks Jeff for clarifying. So closing it.

closed 2022-11-18 by igaw