linux-nvme/nvme-cli #1734: static build: libnvme-mi depends on a shared libsystemd

igaw, 2022-11-17:
libsystemd seems to be only available as shared library. At least on [Debian](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=915566). 

Building against a current version of libnvme with following configuration:

  `CFLAGS=-static meson setup .build --wrap-mode forcefallback --default-library=static `

results in:

```
FAILED: subprojects/libnvme/test/test-mi-mctp 
cc  -o subprojects/libnvme/test/test-mi-mctp subprojects/libnvme/test/test-mi-mctp.p/mi-mctp.c.o subprojects/libnvme/test/test-mi-mctp.p/utils.c.o -Wl,--as-needed -Wl,--no-undefined -Wl,-O1 -static -Wl,--start-group subprojects/libnvme/src/libnvme-mi-test.a subprojects/libnvme/ccan/libccan.a -Wl,--end-group
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: subprojects/libnvme/src/libnvme-mi-test.a.p/nvme_mi-mctp.c.o: in function `nvme_mi_scan_mctp':
mi-mctp.c:(.text+0x804): undefined reference to `sd_bus_default_system'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x84a): undefined reference to `sd_bus_call_method'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x876): undefined reference to `sd_bus_message_enter_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x89a): undefined reference to `sd_bus_message_peek_type'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x8b8): undefined reference to `sd_bus_message_enter_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x8ed): undefined reference to `sd_bus_message_read'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x90b): undefined reference to `sd_bus_message_enter_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x95e): undefined reference to `sd_bus_message_enter_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x975): undefined reference to `sd_bus_message_read_array'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x987): undefined reference to `sd_bus_message_exit_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x9d4): undefined reference to `sd_bus_message_exit_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x9e8): undefined reference to `sd_bus_message_peek_type'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xa04): undefined reference to `sd_bus_message_enter_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xa25): undefined reference to `sd_bus_message_read'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xa6e): undefined reference to `sd_bus_message_peek_type'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xa8a): undefined reference to `sd_bus_message_enter_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xaa8): undefined reference to `sd_bus_message_peek_type'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xad2): undefined reference to `sd_bus_message_enter_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xaec): undefined reference to `sd_bus_message_read'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xb27): undefined reference to `sd_bus_message_read'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xb3b): undefined reference to `sd_bus_message_exit_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xb90): undefined reference to `sd_bus_message_exit_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xbcb): undefined reference to `sd_bus_message_skip'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xbf7): undefined reference to `sd_bus_message_read'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xc13): undefined reference to `sd_bus_message_skip'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xc6a): undefined reference to `sd_bus_message_exit_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xcf3): undefined reference to `sd_bus_message_exit_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xd04): undefined reference to `sd_bus_message_exit_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xd89): undefined reference to `sd_bus_message_exit_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xe20): undefined reference to `sd_bus_error_free'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xe2a): undefined reference to `sd_bus_message_unref'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xe34): undefined reference to `sd_bus_unref'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xf8d): undefined reference to `sd_bus_error_free'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xf97): undefined reference to `sd_bus_message_unref'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0xfa1): undefined reference to `sd_bus_unref'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x107b): undefined reference to `sd_bus_message_exit_container'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x1089): undefined reference to `sd_bus_error_free'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x1093): undefined reference to `sd_bus_message_unref'
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: mi-mctp.c:(.text+0x109d): undefined reference to `sd_bus_unref'
collect2: error: ld returned 1 exit status
```

That begs the question if we should stop trying to get a static build setup running or not.

@jk-ozlabs libsystemd is sucked in via the libnvme-mi library. As I understand we just need the D-Bus API. So I wonder how realistically you see it that we replace it with something else? If that even exists?

jk-ozlabs, 2022-11-18:
A couple of options:

1. Use a non-dbus-enabled libnvme-mi for those builds. The `libsystemd` dependency is optional, and we only use that to query available MI endpoints on the system (which we haven't even plumbed that through to `nvme-cli` yet...).
2. Use a different dbus library - the usage of the dbus API is pretty light, so changing shouldn't be much of an issue.

Either way works for me. Is there a particular use-case for the static builds that might guide this choice?

jk-ozlabs, 2022-11-18:
We could always use the `libdbus` lower-level library, too. Just that:

> If you use this low-level API directly, you're signing up for some pain.

igaw, 2022-11-18:
I also thought that the dependency to libsystemd is optional but my attempt to build nvme-cli statically went horrible wrong. That means something is still not working as expected. 

I'd say we don't need to switch a to different D-Bus binding if it's optional. If someone really wants this feature I accept contributions. 

(I know libdbus very well and it's not so bad as it sounds, it's just a lot of boiler plate code we have to add to make it work)

igaw, 2022-11-18:
This might be related how I was passing the `-static` flag to the compiler. There is a different behavior the environment has `CLFAGS=-static` or one provides it via the `--cross-file` feature.

igaw, 2022-11-18:
FWIW, here is my current draft version of my attempt: https://github.com/igaw/nvme-cli/pull/new/build-add-more-CI

jk-ozlabs, 2022-11-21:
So it looks like having a dynamic `libsystemd` present still allow the libnvme pkg-config check to proceed, and we will happily link the .a with symbols that would still require resolving later:

```
$ objdump -t install.static/lib/x86_64-linux-gnu/libnvme-mi.a | grep UND | grep sd_
0000000000000000         *UND*	0000000000000000 sd_bus_default_system
0000000000000000         *UND*	0000000000000000 sd_bus_call_method
0000000000000000         *UND*	0000000000000000 sd_bus_message_enter_container
0000000000000000         *UND*	0000000000000000 sd_bus_message_peek_type
0000000000000000         *UND*	0000000000000000 sd_bus_message_read
0000000000000000         *UND*	0000000000000000 sd_bus_message_read_array
0000000000000000         *UND*	0000000000000000 sd_bus_message_exit_container
0000000000000000         *UND*	0000000000000000 sd_bus_message_skip
0000000000000000         *UND*	0000000000000000 sd_bus_error_free
0000000000000000         *UND*	0000000000000000 sd_bus_message_unref
0000000000000000         *UND*	0000000000000000 sd_bus_unref
```

(this isn't specific to systemd; the same applies for libc symbols, for example).

This isn't necessarily an invalid configuration, as those symbols may be later resolvable - either as a later link stage, or at runtime. For an eventual static binary though, we probably want all of those to be resolved, and cannot do at a later link stage, because there is no static libsystemd.

So, we probably want some way to explicitly disable the libsystemd dependency, and require that for a static nvme-cli. Manually disabling a dependency seems to be discouraged (see https://github.com/mesonbuild/meson/issues/8224), so instead we may need a feature option. Perhaps a new option for libnvme:

```
 -DDBUS=no
```

\- which we would apply to static builds of nvme-cli?

jk-ozlabs, 2022-11-21:
... which is still a little ugly, as it somewhat complicates the user-interface to the libnvme build system. Also `--default-library=both` will not support dbus, even in the shared lib.

So maybe the best approach would be to switch to libdbus after all?

jk-ozlabs, 2022-11-21:
(are you linking libc statically too?)

igaw, 2022-11-21:
What about making the discovery feature which depends on systemd optional? This would be way cleaner in my opinion.

jk-ozlabs, 2022-11-21:
yep, it already is - but we would need an explicit option to disable it even if libsystemd is found (because the libsystemd that does get found is unsuitable to link against).

I've already done a patch to convert to `libdbus` though, just chasing down a memory leak. Shouldn't be too much more work!

igaw, 2022-11-21:
My idea was to use the option to make the dependency explicit. So if the user wants to use it we look for libsystemd and fail if we can't link. The static build for nvme-cli would just disable it and we should be fine.

But if you got libdbus working, than this is also cool :)

jk-ozlabs, 2022-11-21:
> My idea was to use the option to make the dependency explicit.

Not sure what you mean there - the dependency is already explicit in the `libnvme` meson.build? Is there some trickery we should be doing in the `nvme-cli` meson rules?

igaw, 2022-11-22:
https://gitlab.freedesktop.org/dbus/dbus/-/merge_requests/368

igaw, 2022-11-23:
Let's keep this issue open until we have figured out how to pull in libdbus via subprojects.

igaw, 2022-11-23:
On the second thought. No let's create a new issue for this :)

closed 2022-11-23 by igaw