linux-nvme/nvme-cli #1739: nvme-list produces error log entry each time (Samsung 950 Pro)

dswarbrick, 2022-12-04:
nvme-cli 2.2.x (and possibly earlier 2.x versions) results in an error log entry being created whenever the `nvme list` command is executed, in a system with a Samsung 950 Pro. Version 1.16 did **not** exhibit this.

Drive firmware is 2B0QBXX7 (latest available), and according to smartctl, the supported NVMe version is < 1.2.

The produced error log entries are always the same:
```
# nvme error-log -e 2 /dev/nvme0
Error Log Entries for device:nvme0 entries:2
.................
 Entry[ 0]   
.................
error_count     : 200
sqid            : 0
cmdid           : 0xa00c
status_field    : 0x2002(INVALID_FIELD: A reserved coded value or an unsupported value in a defined field)
phase_tag       : 0
parm_err_loc    : 0
lba             : 0
nsid            : 0x1
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
cs              : 0
trtype_spec_info: 0
.................
 Entry[ 1]   
.................
error_count     : 199
sqid            : 0
cmdid           : 0x200a
status_field    : 0x2002(INVALID_FIELD: A reserved coded value or an unsupported value in a defined field)
phase_tag       : 0
parm_err_loc    : 0
lba             : 0
nsid            : 0x1
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
cs              : 0
trtype_spec_info: 0
.................
```
Since this system runs a Prometheus node_exporter textfile collector (https://github.com/prometheus-community/node-exporter-textfile-collector-scripts/blob/master/nvme_metrics.sh), which is executed at regular intervals, the `nvme list` command within that script will continually produce error log entries.

keithbusch, 2022-12-04:
These are occuring from attempting optional identify commands that we have no spec defined way if knowing if they're supported without trying. The errors are harmless and it's unfortunate some drives decided to add entries for these in the error logs.

As for the kernel message spam, we've fixed that up in 6.1, and I believe has been ported to the most current 6.0 stable.

dswarbrick, 2022-12-07:
@keithbusch Thank you for the clarification. I guess I will need to find some way of muting / ignoring the ever-increasing error log counter on this particular model.

igaw, 2022-12-07:
- kernel error log fix: https://github.com/torvalds/linux/commit/d7ac8dca938cd60cf7bd9a89a229a173c6bcba87
@jk-ozlabs introduced a quirk framework for libnvme-mi to address some fallouts with the MI interface.  We might want to do add such a framework also for libnvme. But this is for the next major version aka libnvme 2.0