linux-nvme/nvme-cli #1741: How to increase admin connection in nvme connect-all command or connect?

Sayoojntly, 2022-12-15:
Need the command changes to increase number of admin connection to increase  nvme over fabrics.

is there any way to increase the admin connections? 
nvme connect-all -t rdma -a 192.168.0.44 -i 16 -q host2 -s 4420

igaw, 2022-12-19:
The spec defines one Admin Queue per controller. I am not sure if I understand your question correctly though. What is the use case you want to get working?

Sayoojntly, 2022-12-19:
Just want to know, if is there any possibility of increasing the number of admin queues for the drive. As of now 1 admin queue i am getting.

igaw, 2022-12-19:
No, this is not possible. There is only one Admin Queue per controller. Could you explain why you want to have more than one queue? What is the problem you try to solve?

closed 2022-12-22 by Sayoojntly