linux-nvme/nvme-cli #1745: Display Terabytes written in for data_units_read and data_units_written (nvme smart-log -H)

jmhands, 2022-12-26:
enhancement
When executing `nvme smart-log` with `-H`, display TB (terabytes) alongside the raw output. Vendors specify endurance rating in TBW, and NVMe raw output is a bit confusing without looking at the specification.

`sudo nvme smart-log -H /dev/nvme0n1`
example
```
data_units_read                         : 1,526,808,040
data_units_written                      : 1,277,863,316
```

example with proposed enhancement (similar to smartctl output)

```
Data Units Read:                    1,526,808,040 [781 TB]
Data Units Written:                 1,277,863,316 [654 TB]
```

per Figure 207: SMART / Health Information Log Page in NVM-Express-Base-Specification-2.0c-2022.10.04-Ratified.pdf

**Data Units Read**: Contains the number of 512 byte data units the host has read from the
controller as part of processing a SMART Data Units Read Command; this value does not
include metadata. This value is reported in thousands (i.e., a value of 1 corresponds to 1,000
units of 512 bytes read) and is rounded up (e.g., one indicates that the number of 512 byte
data units read is from 1 to 1,000, three indicates that the number of 512 byte data units read
is from 2,001 to 3,000).
Refer to the specific I/O Command Set specification for the list of SMART Data Units Read
Commands that affect this field.
A value of 0h in this field indicates that the number of SMART Data Units Read is not reported.

**Data Units Written**: Contains the number of 512 byte data units the host has written to the
controller as part of processing a User Data Out Command; this value does not include
metadata. This value is reported in thousands (i.e., a value of 1 corresponds to 1,000 units of
512 bytes written) and is rounded up (e.g., one indicates that the number of 512 byte data
units written is from 1 to 1,000, three indicates that the number of 512 byte data units written
is from 2,001 to 3,000).
Refer to the specific I/O Command Set specification for the list of User Data Out Commands
that affect this field.
A value of 0h in this field indicates that the number of Data Units Written is not reported

jmhands, 2023-01-02:
excellent! thanks all, looking to rewrite the NVMe-cli guide on nvmexpresss.org at some point soon, and making things more user-friendly (don't always have to look at NVMe spec) will be incredibly helpful

igaw, 2023-01-20:
Just merged @ikegami-t patch.

igaw, 2023-01-20:
with `LC_ALL=de_DE.UTF-8`

```
Data Units Read                         : 23.181.088 (11,87 TB)
Data Units Written                      : 38.335.749 (19,63 TB)
```

and `unset LC_ALL`

```
Data Units Read                         : 23181088 (11.87 TB)
Data Units Written                      : 38335805 (19.63 TB)
```

closed 2023-01-20 by igaw