linux-nvme/nvme-cli #1746: Sanitize help improvements

jmhands, 2022-12-26:
Display sanitize actions and number examples in the `nvme sanitize` help output

current output of `nvme sanitize` 

```
$ sudo nvme sanitize -h
Usage: nvme sanitize <device> [OPTIONS]

Send a sanitize command.

Options:
  [  --no-dealloc, -d ]                 --- No deallocate after sanitize.
  [  --oipbp, -i ]                      --- Overwrite invert pattern between
                                            passes.
  [  --owpass=<NUM>, -n <NUM> ]         --- Overwrite pass count.
  [  --ause, -u ]                       --- Allow unrestricted sanitize exit.
  [  --sanact=<NUM>, -a <NUM> ]         --- Sanitize action.
  [  --ovrpat=<NUM>, -p <NUM> ]         --- Overwrite pattern.
```

proposed output of `nvme sanitize` 

```
$ sudo nvme sanitize -h
Usage: nvme sanitize <device> [OPTIONS]

Send a sanitize command.

Options:
  [  --no-dealloc, -d ]                 --- No deallocate after sanitize.
  [  --oipbp, -i ]                      --- Overwrite invert pattern between
                                            passes.
  [  --owpass=<NUM>, -n <NUM> ]         --- Overwrite pass count.
  [  --ause, -u ]                       --- Allow unrestricted sanitize exit.
  [  --sanact=<NUM>, -a <NUM> ]         --- Sanitize action: 2 = Block Erase, 3 = Overwrite , 4 = Crypto Erase
  [  --ovrpat=<NUM>, -p <NUM> ]         --- Overwrite pattern.
```

ikegami-t, 2023-01-21:
The PR #1748 is already merged into master then the sanitize help is changed as below so seems the issue can be closed.
```
tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme sanitize --help
Usage: nvme sanitize <device> [OPTIONS]

Send a sanitize command.

Options:
...
  [  --sanact=<STRING>, -a <STRING> ]   --- Sanitize action: 1 = Exit failure
                                            mode, 2 = Start block erase, 3 =
                                            Start overwrite, 4 = Start
                                            crypto erase
...
```

igaw, 2023-01-22:
Alright let's close it then.

closed 2023-01-22 by igaw