linux-nvme/nvme-cli #1747: nvme create-ns input of --nsze adn --ncap in standard SI units

jmhands, 2022-12-26:
accept SI units in `--nsze` and `--ncap` for `nvme create-ns`

```
  [  --nsze=<IONUM>, -s <IONUM> ]       --- size of ns (NSZE)
  [  --ncap=<IONUM>, -c <IONUM> ]       --- capacity of ns (NCAP)
```

example current command
`nvme create-ns /dev/nvme0 –nsze 11995709440 –ncap 1199570940 –flbas 0 –dps 0 –nmic 0`

proposed command
`nvme create-ns /dev/nvme0 –nsze 6.14T –ncap 6.14T –flbas 0 –dps 0 –nmic 0`

ikegami-t, 2022-12-28:
FYI: Just confirmed as currenty the binaly units can be used as below but not SI units then the value is just parsed to 6,597,069,766,656 (= 6 x 1,024 x 1,024 x 1,024 x 1,024) as not divided by the LBA size.
```
nvme create-ns /dev/nvme0 -nsze 6Ti -ncap 1199570940 -flbas 0 -dps
```
Note: `6T` is also can be used insed but the prased value is same with `6Ti`. Also `6.14Ti` decimal value using `.` is not parsed correctly as parsed to 0.

igaw, 2023-01-03:
The change is merged.

closed 2023-01-03 by igaw