linux-nvme/nvme-cli #1752: We are should check input of command "nvme list < devices>"

nguoido, 2022-12-30:
I run help for more information on a specific command
```
sudo nvme help
nvme-1.9
usage: nvme <command> [<device>] [<args>]

```

Test case: **sudo nvme list a**
with "a" is the device path
I expect this command will print the error. **But it runs ok**
I think that we should fix it to clear

keithbusch, 2022-12-30:
Hm, I think it's fine. Typical gnu parsing rules let the program stop processing non-argument options (ones that don't start with a '-') whenever they want. The `list` command doesn't interpret such parameters, so ignoring them is aligned with programs that do similarly.

Consider the program `uptime`. Now type `uptime foobar`. The extraneous param is ignored, and program runs ok. But try `uptime --foobar` and you'll get an error.

nguoido, 2022-12-30:
How to get nvme list for one device? I don't want get all nvme devices.

keithbusch, 2022-12-30:
If you already know the device you want to list, you could just run `id-ctrl` on it and get most of the same info `list` tells you. The rest you should be able to get `list-subsys <device>`.

If anyone really thinks the output format of the list command is useful for a single device, though, I suppose that's easy enough to support.

nguoido, 2022-12-30:
I need the value of SectorSize. But I don't see lba value from the output of id-ctrl.
`json_object_add_value_int(jdevice, "SectorSize", lba);`
If you know this, please help me.

ikegami-t, 2023-01-01:
> I need the value of SectorSize. But I don't see lba value from the output of id-ctrl. `json_object_add_value_int(jdevice, "SectorSize", lba);` If you know this, please help me.

The namespace LBA size can be confirmed by the id-ns command flbas and lbaf values as below. (In this case the size is 512 bytes.)
```
nvme id-ns /dev/nvme0n1 -H
NVME Identify Namespace 1:
...
flbas   : 0
  [6:5] : 0     Most significant 2 bits of Current LBA Format Selected
  [4:4] : 0     Metadata Transferred in Separate Contiguous Buffer
  [3:0] : 0     Least significant 4 bits of Current LBA Format Selected
...
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0 Best (in use)
```

nguoido, 2023-01-01:
Thanks you very much. I am clean so I close the ticket

nguoido, 2023-01-01:
Thanks you very much. I am clean

closed 2023-01-02 by igaw