linux-nvme/nvme-cli #1754: Post Linux builds in releases

jmhands, 2023-01-03:
Suggestion to post .deb and .rpm builds as part of the release assets. Ubuntu / Debian / Proxmox are all still on nvme-cli version 1.12.

jmhands, 2023-01-03:
I went through the entire build process a few months ago, which was pretty painful. I will give it another shot, as I see there were some updates to the build process. Builds posted in the releases would still be a nice add, though!

igaw, 2023-01-03:
I wont stop anyone from going through the hassle to provide backport packages for the various distributions. But I don't see this as something I am going to work on. So if no one steps up for this tasks it is unlikely to happen.

keithbusch, 2023-01-03:
I'm also out of the loop on the distro package update requests.

igaw, 2023-01-03:
FWIW, I am working on getting a static builds for nvme-cli/libnvme done (still a few things missing but we are not far away). When it is finally working we could build nightly builds and upload it somewhere (if we get someone to sponsor upload capacity).

jmhands, 2023-01-03:
> I'm also out of the loop on the distro package update requests.

having the latest code is good, especially for all the plugins. I don't know the process for getting the distros to update, but could certainly investigate

igaw, 2023-01-24:
The AppImage approach could be provide solution for this problem. The only thing which seems a bit awkward is where to put the latest generated image. github only allows to generate stable URLs when doing releases. Releases are only possible with a tag. 

That means for sharing via github we would need update the 'latest' tag constantly. But I really don't want the workflow to have the rights to do this.  

An alternative is just to create a new repo which stores the resulting binary. Also a bit hacky.

Best thing would if we had an place to upload a binary and just set the link to it.

jmhands, 2023-03-11:
the 2.3 build compiles fine on Ubuntu 22.04 with kernel 5.15, but on 5.19 I get this error
```nvme: error while loading shared libraries: libnvme.so.1: cannot open shared object file: No such file or directory```

closed 2023-03-14 by igaw