linux-nvme/nvme-cli #1759: `nvme(1)` should get a switch to stay in the foreground

moviuro, 2023-01-06:
# Why?

I'm trying to write a `systemd.unit(5)` for `nvme-device-self-test(1)` but because there is no background/wait-until-complete switch, it is impossible to write a unit that can be properly started and stopped.

# Example

```
# nvme --foreground device-self-test --self-test-code=2 /dev/nvme0n1
... useful data can be printed
... test complete!
#
```

When the nvme command is complete it can even, depending on the command, return an exit status (in the case of device-self-test, 0 if no issues were found, 2 if the test was aborted, etc.)

```ini
[Unit]
Description=Run a short nvme device test on %f
Documentation=man:nvme-device-self-test(1)

[Service]
Type=oneshot
ExecStart=/usr/bin/nvme --foreground device-self-test -s 1 %f
ExecStop=/usr/bin/nvme device-self-test -s 0xf %f
```

keithbusch, 2023-01-06:
I'm unfamiliar with writing such units, but is it possible to do a more complicated script instead? If you can, loop on `nvme self-test-log -e 0 %f | grep "Current Completion"`, and wait for the line to indicate 100%?

I suppose we could also do that within a single command too. The extended self tests can go for several hours, though.

moviuro, 2023-01-09:
@keithbusch this would cause issues with the "real" PID of the service though: what process is in charge of the test, who do we kill/signal when something happens, etc.? The loop would also add unnecessarily "load" to the system for limited gain (though a while sleep loop is not that much, it could add up).

keithbusch, 2023-01-09:
Well, *something* is going to have to do a sleep loop on the self test log, so I'm not sure if there's anything we can do about the load concern.

igaw, 2023-01-09:
I was pondering if the wakeup could be done via udev? Though I have no clue if this makes sense.

keithbusch, 2023-01-09:
If the self test completion was accompanied by AEN, then udev might be okay, but it's not. We can only poll the log, so something will need to take on that responsibility.

igaw, 2023-01-10:
So the idea is that when `nvme device-self-test --self-test-code=1 /dev/nvme0n1 --wait` (or whatever is called), we first check if there is ongoing self test. If so we wait until it finishes in a poll loop or start a new self test and then do the poll loop.

moviuro, 2023-01-10:
Sounds like a good starting point. Further question: what happens when `nvme(1)` gets a signal?

* TERM: stop following/looping or stop the test?
* USR1: print data?
* HUP ?

keithbusch, 2023-01-10:
We're not catching any signals, so the program would just terminate as long as it's not in an uninterruptible state.

igaw, 2023-01-11:
@martin-belanger has reported that reads/writes on `/dev/nvme-fabrics` wont handle signals. So to get signals working we also need to look at this interface.

martin-belanger, 2023-01-11:
I can confirm that `read()` / `write()` will not act as a [cancellation point](https://pubs.opengroup.org/onlinepubs/000095399/functions/xsh_chap02_09.html#tag_02_09_05_02) when operating on `/dev/nvme-fabrics`. Typically, when sending a signal to a process/thread while it is blocked on a `read()` / `write()` will cause the operation to return immediately with `errno==EINTR`. However, that is not the case when the operation is performed on `/dev/nvme-fabrics`. In fact, the same applies to `delete_controller` (e.g. `/sys/class/nvme/nvme0/delete_controller`).

This can become a problem when an application tries to make several connections (> 100) concurrently using threads and all threads are blocked waiting for their turn to be serviced by the kernel. It will be impossible to SIGTERM that process until all the connections have been processed.

igaw, 2023-01-23:
With the current HEAD it's possible to do `nvme device-self-test -s 1 -w`. Hope this helps your projejct!

closed 2023-01-23 by igaw