linux-nvme/nvme-cli #1760: File System Synchronization issue

antosoft, 2023-01-09:
Can someone help me with the file system synchronization issue? When I create a file on the client it doesn't update on target.
I have installed nvme-cli nvme version 2.2.1 (git 2.2.1), libnvme version 1.2 (git 1.2) which I compile from source myself. I have two VMs, the target is a linux debian 11 and the client is a linux rhel9.
On the client I can mount /dev/nvme1n1 just fine after running nvme connect-all. The problem is that any file modification does not synchronize with the target. If it's not an nvme-cli bug, how do you fix this?

igaw, 2023-01-09:
nvme-cli is not involved anymore after the initial nvme-tcp setup `nvme connect-all` (I assume you are running the TCP transport). 

Unfortunately, I can't really say much as you haven't described the setup in more detail. E.g. the nvmet setup etc.

Could you elaborate a bit? Maybe we can point you to the right direction (mailing list)

antosoft, 2023-01-09:
I have a 2TB NVME Gen3 drive attached to my proxmox 7 server running debian 11, partitioned and mounted as follows:

```console
lsblk
nvme0n1      259:0    0   1,8T  0 disk
├─nvme0n1p1  259:1    0   1,5T  0 part /mnt/vmdata
└─nvme0n1p2  259:2    0   327G  0 part /mnt/appdata
```
The `/dev/nvme0n1p1` partition I use for virtual machine data, and the `/dev/nvme0n1p2` I want to use for application data, and I need to mount inside the virtual machines for which I have configured `nvme-cli`.

On my proxmox server with linux debian 11 **(with vlan10@eth0, IP: 10.10.100.2/24)**, which has the physical NVME drive that will be my TARGET, I have the following nvme-cli version installed, which compiles it from source:
```console
nvme version
nvme version 2.2.1 (git 2.2.1)
libnvme version 1.2 (git 1.2)
```

I do the configuration with the following script that is executed when booting the proxmox server:

```bash
TARGET=10.10.100.2  #<--- IP of proxmox and TARGET for nvme-cli
DEV=/dev/nvme0n1p2  #<--- Partition to use with nvme-cli (I have it mounted in /mnt/appdata in proxmox)
PORT=4420
NVME_PORT=1
NQN=nqn.antosoft.net.application-data
NSID=1

if [[ ! -f /sys/kernel/config/nvmet/subsystems/${NQN} ]]; then
    echo creating ${NQN} on device ${DEV}
    modprobe -a configfs nvme
    modprobe -a nvmet nvmet_tcp

    cd /sys/kernel/config/nvmet/subsystems/
    mkdir ${NQN}
    sleep 1

    mkdir ${NQN}/namespaces/${NSID}
    echo -n ${DEV} > ${NQN}/namespaces/${NSID}/device_path
    echo -n 1 > ${NQN}/attr_allow_any_host
    echo -n 1 > ${NQN}/namespaces/${NSID}/enable

    cd /sys/kernel/config/nvmet/ports
    mkdir ${NVME_PORT}
    echo -n ipv4 > ${NVME_PORT}/addr_adrfam
    echo -n tcp > ${NVME_PORT}/addr_trtype
    #echo -n 1 > ${NVME_PORT}/addr_treq
    echo -n ${TARGET} > ${NVME_PORT}/addr_traddr
    echo -n ${PORT} > ${NVME_PORT}/addr_trsvcid

    ln -s ../subsystems/${NQN} ${NVME_PORT}/subsystems/${NQN}
fi
```

Finally, I have port 4420 allowed in the proxmox firewall.

**CLIENT VIRTUAL MACHINE (with vlan10@eth0, IP: 10.10.100.3/24)**
It is a virtual machine inside the proxmox server, with a rhel 9 distro. With the same version of nvme-cli compiled from source, then installed on the system without errors:

```console
nvme version
nvme version 2.2.1 (git 2.2.1)
libnvme version 1.2 (git 1.2)
```
On the client I load the following modules from `/etc/modules-load.d/modules.conf`
```console
nvme_core
nvme_fabrics
nvme
nvme_tcp
```

In the file `/etc/nvme/discovery.conf` I enter the following:
```console
-t tcp -a 10.10.100.2 -s 4420
```

In the firewall I establish the following:
```console
firewall-cmd --new-zone=nvme-access --permanent
firewall-cmd --zone=nvme-access --add-source=10.10.100.0/24 --permanent
firewall-cmd --zone=nvme-access --add-port=4420/tcp --permanent
firewall-cmd --zone=nvme-access --add-rich-rule='rule family="ipv4" source address="10.10.100.3/24" port protocol="tcp" port="4420" accept' --permanent 
firewall-cmd --reload
```

Then I run the following commands:

```console
nvme discover

Discovery Log Number of Records 0, Generation counter 1
=====Discovery Log Entry 0======
trtype:  tcp
adrfam:  ipv4
subtype: nvme subsystem
traddr:  10.10.100.2
treq:    not required
trsvcid: 4420
subnqn:  nqn.antosoft.net.application-data
sectype: none
```

```console
nvme connect-all
```

```console
lsblk

NAME                                        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
sda                                           8:0    0   32G  0 disk
├─sda1                                        8:1    0    1G  0 part /boot
└─sda2                                        8:2    0   31G  0 part
  ├─rhel9_rrcs--172--254--100--217-root 253:0    0 27,8G  0 lvm  /
  └─rhel9_rrcs--172--254--100--217-swap 253:1    0  3,2G  0 lvm  [SWAP]
sr0                                          11:0    1  804M  0 rom
nvme1n1                                      12:0    0  327G  0 disk
```
At this point I mount the new device that nvme-cli connects to me `/dev/nvme1n1` (so far everything is as expected).
```console
mkdir /mnt/app-data
mount /dev/nvme1n1 /mnt/app-data -o rw
```

**HERE IS THE PROBLEM**
Now when I create, modify or delete a file in `/mnt/app-data` those changes are not reflected in the `/mnt/appdata` folder mounted on the proxmox server (TARGET).  The same happens in reverse.

**How should this connection work?**

Because if I use nfs-server for example: Yes, I can create, modify, delete files and the changes are instantly reflected on the other end.

keithbusch, 2023-01-09:
You can't have two users mounting the same drive. Each local filesystem will think it's the only entity that's opened the block device and will soon be out of sync with each other. The nvme target code might need to add an exclusive open flag to prevent this kind of setup.

igaw, 2023-01-09:
If you want to share data between hosts/guests you need a shared filesystem, such as nfs or any of the cluster filesystem. Sharing on block layer is not supported at all. I suppose if you just want read access you could get away with your setup but still not recommend.

igaw, 2023-01-09:
I see where your expectation is coming from. With `mount --bind` it is possible to mount the FS several times. This works because there is only one kernel involved which keeps everything in sync (from user point of view).

In your setup there is more than one kernel involved and these kernels do not synchronize their operations on the physical block device.  

If you want to synchronize on block layer you would need something like DRBD.

closed 2023-01-09 by igaw