linux-nvme/nvme-cli #1763: Is it possible to add firmware reading option?

Nefarious19, 2023-01-12:
### Why?

sometimes, for some nvme SSDs, there is no firmware on manufacturers site. There are only .EXE instalation files from laptop manufacturers like lenovo. I think it would be possible to extract SSD fw from those EXE files. If we could read old fw from SSD we could compare it with EXE hexdump and possibly find region where FW is placed by looking for similarities.

You can buy some used SSDs from laptops in very good condition. But you can't find usually FW for those on manufacturers websites.

This is only idea.

keithbusch, 2023-01-12:
There's no spec defined way to read firmware binary from the drive.

It is frustrating that many vendors wrap their firmware in a proprietary executable, though.

DataDrug, 2023-01-12:
Adding my 2 cents.

I work in data recovery. Nvme is way diferent from sata. We have been having a ton of issues recovering nvme drives with failed controllers and with bad firmware due to the way they are designed. TxRx is usually locked and jtag is also disabled from factory.

Even if you could read the firmware from the drive you would not be able to compare it with the .EXE as it is obfuscated.

If you really wish to follow this path (i advise you not to do it unleas you enjoying banging your head against the wall) you must start by trying to bypass jtag and terminal lock (by using a chipwhisperer for example).

You can also try to intercept the data from the exe into the nvme drive while it is updating, but this is tedious.

However, if you are interested in the nvmes used by lenovo  latops, I can tell you they often use Marvell. If you have access, take a high resolution xray, and you will get the firmware easily if you access some obvious second layer traces.

Good luck.

closed 2023-01-13 by Nefarious19