linux-nvme/nvme-cli #1764: make appimage format or binary file

newyorkthink, 2023-01-16:
inxi bat or other command , they allow have binary file.

appimage or binary they not use apt-get ,it can run every e:)

igaw, 2023-01-16:
If I understand your correctly, you are asking us to provide a nvme-cli packaged up as an AppImage,

- https://github.com/AppImageCrafters/appimage-builder
- https://appimage-builder.readthedocs.io/en/latest/hosted-services/github-actions.html

Recently, we had already a quick discussion if we should provide binaries upstream. My response was that I am not willing to spend time on packaging for distributions, so we rather look into how we can provide statically linked binaries.

Though I haven't really thought about AppImage. If it is really just setting up a workflow on github and we are done with it, I am fine to look into.

igaw, 2023-01-16:
See also #1754

igaw, 2023-01-20:
We still need to figure out where to upload the result. Currently the image is only available via the artifacts attached to the latest build.

newyorkthink, 2023-01-20:
I success build :)

newyorkthink, 2023-01-20:
[image]

igaw, 2023-01-24:
I've experimented with uploading the generated AppImage as release artifact. To get this working the workflow needs to create or update a tag (e.g. tip or latest). This is really ugly and I don't want to allow the workflow actually to do the tags.

That means we need some place for uploading the generated artifact unless we are okay with pointing to the build artifacts. Note these don't have a stable URL...

jmhands, 2023-03-08:
I see the appimage in the actions, is it possible to upload the appimage for the release builds to make it easy for people to find?

igaw, 2023-03-09:
We are just missing a place to upload build artifacts.

igaw, 2023-04-03:
There we go: https://monom.org/linux-nvme/

closed 2023-04-03 by igaw

docop, 2023-06-20:
Very usefull to have a true portable system and quick check of nvme with this true portable version in appimage. great.

PR #1769 (2023-01-20): Create AppImage
Fixes #1764