linux-nvme/nvme-cli #1766: Error output is split between stderr and stdout

mwilck, 2023-01-17:
Current openSUSE Tumbleweed:
```
# nvme --version
nvme version 2.2.1 (git 2.2.1)
libnvme version 1.2 (git 1.2)

# nvme invalid-command >/tmp/stdout.txt 2>/tmp/stderr.txt
# wc /tmp/std*.txt
   2   25  169 /tmp/stderr.txt
 135  908 9038 /tmp/stdout.txt

# cat /tmp/stdout.txt
ERROR: Invalid sub-command 'invalid-command'
nvme-2.2.1
usage: nvme <command> [<device>] [<args>]

The following are all implemented sub-commands:
  list                         List all NVMe devices and namespaces on machine
  list-subsys                  List nvme subsystems
  id-ctrl                      Send NVMe Identify Controller
  id-ns                        Send NVMe Identify Namespace, display structure
...
  help                         Display this help

See 'nvme help <command>' for more information on a specific command

The following are all installed plugin extensions:
  ocp             OCP cloud SSD extensions
  zns             Zoned Namespace Command Set
...
  amzn            Amazon vendor specific extensions

See 'nvme <plugin> help' for more information on a plugin

# cat /tmp/stderr.txt 
The '<device>' may be either an NVMe character device (ex: /dev/nvme0), an
nvme block device (ex: /dev/nvme0n1), or a mctp address in the form
mctp:<net>,<eid>[:ctrl-id]```

Some output goes to stderr, but most goes to stdout. This is suboptimal, given that shell scripts will likely do things like
```
nvme fancy-command 2>/dev/null || FANCY_UNSUPPORTED
```
(which will print lots of undesirable garbage), and sort of violates the established standards for Unix command line utilities.

IMO output like this should go to stderr exclusively.

ikegami-t, 2023-01-22:
* `general_help()` uses `printf()`, `usage()` and `print_word_wrapped()` to output help messages on the error case as below.
```
void general_help(struct plugin *plugin)
{
...
	printf("%s-%s\n", prog->name, prog->version);

	usage(plugin);

	printf("\n");
	print_word_wrapped(prog->desc, 0, 0);
	printf("\n");
...
```
* `printf()` outputs into `stdout` but `print_word_wrapped()` uses `putc` as outputs into `stderr`.
```
void print_word_wrapped(const char *s, int indent, int start)
{
...
	while (start < indent) {
		putc(' ', stderr);
...
	}

	for (c = s; *c != 0; c++) {
...
			if (((int)(c - s) + start + next_space) > (last_line - indent + width)) {
...
				putc('\n', stderr);
				for (i = 0; i < indent; i++)
					putc(' ', stderr);
...
			}
		}
		putc(*c, stderr);
	}
}
```
* `usage()` also uses `printf()` so outputs into `stdout`.
```
void usage(struct plugin *plugin)
{
...
	if (plugin->name)
		printf("usage: %s %s %s\n", prog->name, plugin->name, prog->usage);
	else
		printf("usage: %s %s\n", prog->name, prog->usage);
}
```

igaw, 2023-01-24:
Just merged @ikegami-t fix.

closed 2023-01-24 by igaw