linux-nvme/nvme-cli #1767: Error (status 0x2002) when running `nvme list`

Bacto, 2023-01-18:
Hi,

Each time I'm using `nvme list`, the error log get a new entry of this kind:
```
error_count	: 1198
sqid		: 0
cmdid		: 0xb004
status_field	: 0x2002(Invalid Field in Command: A reserved coded value or an unsupported value in a defined field)
phase_tag	: 0
parm_err_loc	: 0xffff
lba		: 0
nsid		: 0x1
vs		: 0
trtype		: The transport type is not indicated or the error is not transport related.
cs		: 0
trtype_spec_info: 0
```

I have this on 2 NVMe disks SAMSUNG MZQLB960HAJR-00007 (firmware EDA5702Q).

```
# nvme id-ctrl /dev/nvme0
NVME Identify Controller:
vid       : 0x144d
ssvid     : 0x144d
sn        : <hidden>
mn        : SAMSUNG MZQLB960HAJR-00007
fr        : EDA5702Q
rab       : 2
ieee      : 002538
cmic      : 0
mdts      : 9
cntlid    : 0x4
ver       : 0x10200
rtd3r     : 0x7a1200
rtd3e     : 0x7a1200
oaes      : 0
ctratt    : 0
rrls      : 0
cntrltype : 0
fguid     :
crdt1     : 0
crdt2     : 0
crdt3     : 0
nvmsr     : 0
vwci      : 0
mec       : 0
oacs      : 0xf
acl       : 7
aerl      : 3
frmw      : 0x17
lpa       : 0x3
elpe      : 63
npss      : 0
avscc     : 0x1
apsta     : 0
wctemp    : 360
cctemp    : 361
mtfa      : 0
hmpre     : 0
hmmin     : 0
tnvmcap   : 960197124096
unvmcap   : 0
rpmbs     : 0
edstt     : 0
dsto      : 0
fwug      : 0
kas       : 0
hctma     : 0
mntmt     : 0
mxtmt     : 0
sanicap   : 0
hmminds   : 0
hmmaxd    : 0
nsetidmax : 0
endgidmax : 0
anatt     : 0
anacap    : 0
anagrpmax : 0
nanagrpid : 0
pels      : 0
domainid  : 0
megcap    : 0
sqes      : 0x66
cqes      : 0x44
maxcmd    : 0
nn        : 1
oncs      : 0x1f
fuses     : 0
fna       : 0x4
vwc       : 0
awun      : 1023
awupf     : 7
icsvscc     : 1
nwpc      : 0
acwu      : 0
ocfs      : 0
sgls      : 0
mnan      : 0
maxdna    : 0
maxcna    : 0
subnqn    :
ioccsz    : 0
iorcsz    : 0
icdoff    : 0
fcatt     : 0
msdbd     : 0
ofcs      : 0
ps    0 : mp:10.60W operational enlat:0 exlat:0 rrt:0 rrl:0
          rwt:0 rwl:0 idle_power:- active_power:-
```

For what I understood this error is because `nvme list` uses an unknown command. So it's not a big problem but it's annoying to see the error log being filled by this.

Tested with nvme-cli 2.0 and 2.2.1.

Best,
Adrien

> By the way, is there any way to erase the error log?

igaw, 2023-01-18:
Known problem with the Samsung firmwares: https://github.com/linux-nvme/libnvme/issues/550

igaw, 2023-01-18:
No problem. I hope we can address this eventually. As described in the other issue, the idea is to introduce a quirk framework for such cases. 

BTW, I don't know if there is a way to reset the error log.