linux-nvme/nvme-cli #1785: nvme-list listing 2 controllers though ns is attached from only one controller.

sarabjotsoit, 2023-01-30:
Steps to check:

1. Connect the drives from 2 paths(controllers).
 2. Create a namespace on the drive and attach it from one path.
 3. check "nvme list -v".

Actual result:

It lists 2 controllers though ns is attached from only one controller.
{noformat}
root@dcs:~# nvme list -v
Subsystem        Subsystem-NQN                                                                                    Controllers
---------------- ------------------------------------------------------------------------------------------------ ----------------
nvme-subsys9     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.9                      nvme9, nvme33
nvme-subsys8     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.8                      nvme8, nvme32
nvme-subsys7     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.7                      nvme7, nvme31
nvme-subsys6     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.6                      nvme6, nvme30
nvme-subsys5     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.5                      nvme5, nvme29
nvme-subsys4     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.4                      nvme4, nvme28
nvme-subsys3     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.3                      nvme3, nvme27
nvme-subsys24    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.24                     nvme48, nvme24
nvme-subsys23    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.23                     nvme47, nvme23
nvme-subsys22    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.22                     nvme46, nvme22
nvme-subsys21    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.21                     nvme45, nvme21
nvme-subsys20    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.20                     nvme44, nvme20
nvme-subsys2     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.2                      nvme26, nvme2
nvme-subsys19    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.19                     nvme43, nvme19
nvme-subsys18    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.18                     nvme42, nvme18
nvme-subsys17    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.17                     nvme41, nvme17
nvme-subsys16    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.16                     nvme40, nvme16
nvme-subsys15    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.15                     nvme39, nvme15
nvme-subsys14    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.14                     nvme38, nvme14
nvme-subsys13    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.13                     nvme37, nvme13
nvme-subsys12    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.12                     nvme36, nvme12
nvme-subsys11    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.11                     nvme35, nvme11
nvme-subsys10    nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.10                     nvme34, nvme10
nvme-subsys1     nqn.1992-05.com.wdc.mat-22222222222222222222222222222222222222222222:nvme.1                      nvme25, nvme1
 
Device   SN                   MN                                       FR       TxPort Address        Subsystem    Namespaces
-------- -------------------- ---------------------------------------- -------- ------ -------------- ------------ ----------------
nvme9    A05D3C6B             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys9 nvme9n1
nvme33   A05D3C6B             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys9
nvme8    A06A4A6D             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys8
nvme32   A06A4A6D             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys8
nvme7    A05D3BC8             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys7 nvme7n1
nvme31   A05D3BC8             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys7
nvme6    A06A4A73             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys6 nvme6n1
nvme30   A06A4A73             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys6
nvme5    A05D3C22             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys5 nvme5n1
nvme29   A05D3C22             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys5
nvme4    A05D3CF5             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys4 nvme4n1
nvme28   A05D3CF5             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys4
nvme3    A0735EED             WUS4BA176DSP3X4                          R2210400 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys3 nvme3n1
nvme27   A0735EED             WUS4BA176DSP3X4                          R2210400 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys3
nvme48   A06A4B64             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys24
nvme24   A06A4B64             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys24 nvme24n1
nvme47   A06A4ABE             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys23
nvme23   A06A4ABE             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys23 nvme23n1
nvme46   A0735E8E             WUS4BA176DSP3X4                          R2210400 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys22
nvme22   A0735E8E             WUS4BA176DSP3X4                          R2210400 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys22 nvme22n1
nvme45   A073612E             WUS4BA176DSP3X4                          R2210400 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys21
nvme21   A073612E             WUS4BA176DSP3X4                          R2210400 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys21 nvme21n1
nvme44   A06A4A8A             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys20
nvme20   A06A4A8A             WUS4BA1A1DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys20 nvme20n1
nvme26   A06AC147             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys2
nvme2    A06AC147             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys2 nvme2n1
nvme43   A06AC191             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys19
nvme19   A06AC191             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys19 nvme19n1
nvme42   A06ABFEA             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.1.51,trsvcid=4420 nvme-subsys18
nvme18   A06ABFEA             WUS4BA138DSP3X1                          R2210010 rdma   traddr=192.168.0.50,trsvcid=4420 nvme-subsys18 nvme18n1
[... 141 lines cut ...]
/dev/nvme5n1 /dev/ng5n1   1          3.84  TB /   3.84  TB      4 KiB +  0 B   nvme5, nvme29
/dev/nvme4n1 /dev/ng4n1   1          3.84  TB /   3.84  TB      4 KiB +  0 B   nvme4, nvme28
/dev/nvme3n1 /dev/ng3n1   1          7.68  TB /   7.68  TB      4 KiB +  0 B   nvme3, nvme27
/dev/nvme24n1 /dev/ng24n1  1         15.36  TB /  15.36  TB      4 KiB +  0 B   nvme48, nvme24
[... cut ...]
/dev/nvme13n1 /dev/ng13n1  1          7.68  TB /   7.68  TB      4 KiB +  0 B   nvme37, nvme13
/dev/nvme12n1 /dev/ng12n1  1         15.36  TB /  15.36  TB      4 KiB +  0 B   nvme36, nvme12
/dev/nvme11n1 /dev/ng11n1  1          7.68  TB /   7.68  TB      4 KiB +  0 B   nvme35, nvme11
/dev/nvme10n1 /dev/ng10n1  1         15.36  TB /  15.36  TB      4 KiB +  0 B   nvme34, nvme10
/dev/nvme1n1 /dev/ng1n1   1          7.68  TB /   7.68  TB      4 KiB +  0 B   nvme25, nvme1
root@dcs:~#
root@dcs:~# nvme detach-ns /dev/nvme31 -n 1 -c 0x8008
detach-ns: Success, nsid:1

{noformat}

sarabjotsoit, 2023-01-30:
We will provide the kernel logs asap.

igaw, 2023-02-06:
The two `nvme list` output shows that the controller `nvme7` and `nvme31` are attached to the namespace `nvme7n1` before and after the tests. And the order of the commands are 'show, detach, attach, show'.  I don't see what's wrong I must miss something. Could you elaborate? Thanks.

sarabjotsoit, 2023-02-10:
Thank you @igaw 

We are attaching only one of the controllers. This is working as expected in the nvme-cli 1.13/16

igaw, 2023-02-13:
I am confused. Are the logs you provided for nvme-cli 1.x or 2.x? I don't see a problem there.

igaw, 2023-03-01:
No feedback. Closing.

closed 2023-03-01 by igaw