linux-nvme/nvme-cli #1786: nvme version 2.2.1 (git 2.2.1) showing wrong "UsedBytes", "MaximumLBA" and "PhysicalSize"

KiranSL001, 2023-01-31:
In NVME version 2.2.1 our nvme drives are showing wrong values in  "UsedBytes", "MaximumLBA" and "PhysicalSize"
Find  the details below.
[root@SB-Perf-01 ~]# nvme --version
nvme version 2.2.1 (git 2.2.1)
libnvme version 1.2 (git 1.2)
[root@SB-Perf-01 ~]# nvme list -o json
{
  "Devices":[
    {
      "NameSpace":1,
      "DevicePath":"/dev/nvme1n2",
      "Firmware":"LF100010",
      "ModelNumber":"WUS6AA138FSP5EA",
      "SerialNumber":"22482U900075",
      **"UsedBytes":-2147483648,
      "MaximumLBA":-1088458064,
      "PhysicalSize":-2147483648,**
      "SectorSize":512

**With nvme version 1.16 we are getting the correct values.**

[root@SB-Perf-09 ~]# nvme list -o json
{
  "Devices" : [
    {
      "NameSpace" : 1,
      "DevicePath" : "/dev/nvme0n1",
      "Firmware" : "LF100010",
      "Index" : 0,
      "ModelNumber" : "WUS6AA138FSP5EA",
      "ProductName" : "Non-Volatile memory controller: Western Digital Device 0x2704",
      "SerialNumber" : "22482U900075",
      **"UsedBytes" : 3840755982336,
      "MaximumLBA" : 7501476528,
      "PhysicalSize" : 3840755982336,**
      "SectorSize" : 512
    }
  ]
}
[root@SB-Perf-09 ~]# nvme --version
nvme version 1.16

igaw, 2023-01-31:
Should be fixed with 87bf24cdffc1 ("nvme-print: Serialize uint32_t types correctly")

We just released v2.3 which ships this fix. Can you try again with the new release?

KiranSL001, 2023-02-09:
> CAUTION: This email originated from outside of Western Digital. Do not click on links or open attachments unless you recognize the sender and know that the content is safe.
>
> Should be fixed with 87bf24c<https://github.com/linux-nvme/nvme-cli/commit/87bf24cdffc1877002b08464064a2ad642a77317> ("nvme-print: Serialize uint32_t types correctly")

Hi team,

Could you share me the installer file for nvme-cli 2.3?

Make install is not working for me.

Thanks,
Kiran

closed 2023-03-01 by igaw