linux-nvme/nvme-cli #1787: 2.3: test site is failing in `suffix_si_parse` unit

kloczek, 2023-02-01:
Looks like something is wrong.
```console
+ /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 48 --print-errorlogs
ninja: no work to do.
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/nvme-cli-2.3/x86_64-redhat-linux-gnu'
ninja: no work to do.
1/3 uint128         OK              0.02s
2/3 suffix_si_parse FAIL            0.01s   exit status 1
>>> MALLOC_PERTURB_=125 /home/tkloczko/rpmbuild/BUILD/nvme-cli-2.3/x86_64-redhat-linux-gnu/unit/test-suffix-si-parse
――――――――――――――――――――――――――――――――――――― ✀  ―――――――――――――――――――――――――――――――――――――
ERROR: printing {11995709440} (lbas 512), got '23429120', expected '11995709440'
ERROR: printing {1199570940} (lbas 512), got '2342911', expected '1199570940'
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

3/3 uint128-si      OK              0.01s

Summary of Failures:

2/3 suffix_si_parse FAIL            0.01s   exit status 1

Ok:                 2
Expected Fail:      0
Fail:               1
Unexpected Pass:    0
Skipped:            0
Timeout:            0
```
I'm not sure what kind of details I should provide to diagnose that issue.

igaw, 2023-02-01:
@ikegami-t

igaw, 2023-02-01:
So we need something like this:

```diff
@@ -84,9 +84,12 @@ uint64_t suffix_si_parse(const char *value, bool *suffixed)
                }
        }
 
-       if (suffix[0] != '\0')
+       if (suffix[0] != '\0') {
                errno = EINVAL;
+               return 0;
+       }
 
+       *suffixed = false;
        return (uint64_t)ret;
 }
```

But when I look at the test code, I think we have another bug in `suffix_si_parse`:

```c
static struct tonum_test tonum_tests[] = {
	{ "11995709440", 512, 11995709440 },
	{ "1199570940", 512, 1199570940 },
	{ "6.14T", 512, 11992187500 },
	{ "6.14T", 520, 11807692307 },
	{ "6.14T", 4096, 1499023437 },
	{ "6.14", 512, 0 },
	{ "6.14#", 512, 0 },
};
```
The second last entry should not be `0`, instead it should be `6`, no?

igaw, 2023-02-01:
And looking more at it, we also do not take into account, that the string given might be locale-dependent.

igaw, 2023-02-01:
@kloczek what are you locale settings?

```shell
$ env | grep LC_
LC_MEASUREMENT=C
LC_MONETARY=C
LC_NUMERIC=C
LC_TIME=C
LC_ALL=en_US.UTF-8
```

kloczek, 2023-02-02:
> @kloczek what are you locale settings?

As I'm buiding everything as rpm packages rpmbuild during the build always forces locale `C`.

igaw, 2023-02-02:
Right, so this is not causing troubles. But we still have to handle the locales.

kloczek, 2023-02-02:
Just tested https://github.com/linux-nvme/nvme-cli/pull/1788 and looks like it dolves the issue.

igaw, 2023-02-03:
> Just tested #1788 and looks like it dolves the issue.

This patch is good for addressing the issue. Are you fine just adding it to your build chain?

kloczek, 2023-02-03:
I've integrated that into my package to test it😋 
So decision is yours are you happy wit it or not 😃

igaw, 2023-02-03:
I'll apply my re implementation version. Obviously, we could do a v2.3.1 but for this thing alone it's properly overkill, especially if you can add the fix as patch to the build process.

kloczek, 2023-02-03:
Yep 👍 
Feel free to close this ticket 😋 
And .. Thank you 😄

igaw, 2023-02-07:
#1790 is merged, thus the master branch should be fixed... (famous last words).

Thanks for the report!

closed 2023-02-07 by igaw