linux-nvme/nvme-cli #1797: How to retrieve errno code of IO commands in shell?

alexmchou, 2023-02-05:
Hello,
I am trying to test NVMe cli IO commands' error status code in bash script. If I got following error message for a write command: "NVMe status: End-to-End Storage Tag Check Error: The command was aborted due to an end-to-end storage tag check failure(0x288)", I tried to use "echo $?",  but got 1 instead of 136 (0x88).  From the nvme.c code, err (not errno)  is returned by IO command handler. This is different than what I got from history issue #862. Any suggestions?
Thanks

ikegami-t, 2023-02-05:
As below currently only returned 0 on success and 1 on error that was changed by the commit 13f74d2.
```
int main(int argc, char **argv)
{
	int err;

	nvme.extensions->parent = &nvme;
	if (argc < 2) {
		general_help(&builtin);
		return 0;
	}
	setlocale(LC_ALL, "");

	err = handle_plugin(argc - 1, &argv[1], nvme.extensions);
	if (err == -ENOTTY)
		general_help(&builtin);

	return err ? 1 : 0;
}
```

igaw, 2023-02-06:
I think for the normal shell output keeping I think we should leave it as it is (see reasons in https://github.com/linux-nvme/nvme-cli/commit/13f74d20e1d9e571368b1d06fe9889a2fdd29849). This should be fine as it is thought to be used by humans which are able to read/parse the output.

Though I think we could make it for scripts and tooling around nvme a bit more useful which is supposed to use the JSON output. hat is when a command fails we return 1 but also a JSON formatted output with the error code and error message. What do you think of this idea?

ikegami-t, 2023-02-06:
Yes I can agree with your idea.

ikegami-t, 2023-04-01:
By the PR #1876 merged into maser a new `json` parameter has been added as below and then the errno code is able to be retrieved by the `value` output as below for example. Can we close the issue now?
```
[2023-03-25 12:16:14.221] tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme get-feature /dev/nvme0 -f 0xc0 --json
[2023-03-25 12:16:17.768] {
[2023-03-25 12:16:17.768]   "error":"Invalid Field in Command: A reserved coded value or an unsupported value in a defined field",
[2023-03-25 12:16:17.768]   "type":"nvme",
[2023-03-25 12:16:17.768]   "value":8194
[2023-03-25 12:16:17.768] }
```
Note: The `value` parameter is output as signed integer value but not as unsinged hexadecimal value. Also it will be changed to introduce a plugin interface as mentioned by the PR https://github.com/linux-nvme/nvme-cli/pull/1876#issuecomment-1484809888.

igaw, 2023-04-03:
Thanks @ikegami-t !