linux-nvme/nvme-cli #1804: not able to install nvme-cli2.3

KiranSL, 2023-02-09:
make install is not working..
[image]

Requesting provide the help in installing and provide zip file to install in centos

igaw, 2023-02-09:
We have AppImage for such cases. The only problem they are bit hidden. You find them as build artifcats in the CI runs. The urls are not stable so we can't really point to 'latest'. (will be solved eventually.

But try this here:

https://github.com/linux-nvme/nvme-cli/suites/10876523428/artifacts/548771620

closed 2023-02-09 by igaw

KiranSL, 2023-02-09:
Hi Daniel,

While instaling this file i am getting

***@***.*** ~]# ./nvme-cli-latest-x86_64.AppImage
/tmp/.mount_nvme-cSoHXVV/usr/sbin/nvme: error while loading shared
libraries: libcrypto.so.3: cannot open shared object file: No such file or
directory

igaw, 2023-02-10:
Fixed the AppImage build recipe. It includes now libcrypto.so.3  too:

https://github.com/linux-nvme/nvme-cli/suites/10903326839/artifacts/550736621

PR #1809 (2023-02-10): build: Update library paths for AppImage
Ubuntu uses a multilib setup hence the libraries are not found in /lib64.

Fixes: #1804