linux-nvme/nvme-cli #1806: udev rules could be simplified

martin-belanger, 2023-02-09:
I submit this issue for your review. Please let me know if you agree with the following analysis and I will submit a pull request if you do.

The udev rules specified in [70-nvmf-autoconnect.rules.in](https://github.com/linux-nvme/nvme-cli/blob/master/nvmf-autoconnect/udev-rules/70-nvmf-autoconnect.rules.in) are used to start a oneshot service defined by [nvmf-connect%40.service.in](https://github.com/linux-nvme/nvme-cli/blob/master/nvmf-autoconnect/systemd/nvmf-connect%40.service.in). This oneshot service simply invokes the command `nvme connect-all`.

The command `nvme connect-all` supports a `--device` option. When that option is specified, it is not necessary to provide all the other options like `--transport`, `--traddr`, etc. In fact, when `--device` is specified the other options are ignored.

The udev rules are currently passing not only `--device` but also all the other  options to `nvme connect-all`, which is redundant.  

I propose to only use the `--device` option in the udev rules. Let me know what you think.

igaw, 2023-02-10:
One thing which needs to be double checked is if  `connect` and `connect-all` actually do the same thing. If I got this right then `connect-all` will try to use the arguments from the command line and does a full discovery. Anything it finds it will try to connect. Where as  `connect` only does only connect to one controller specified from the command line.  Obviously, if the arguments handed in just a controller than it should be equivalent.

martin-belanger, 2023-02-10:
You are correct, `connect-all` connects to a Discovery Controller (DC), performs a discovery command to retrieve the Discovery Log Pages (DLP), and then connects to every I/O controller listed in the DLP. The `connect-all` command can be invoked by specifying an existing persistent DC connection, in which case one need only provide the `--device` option. And in this case, the options `--transport`, `--traddr`, `--trsvcid` are redundant. 

I need to take a closer look at the code because I think that `--host-traddr` and `--host-iface` are needed because they get applied to the I/O controller connections. However, these two options could be retrieved from the existing DC connection, which means that when invoking `connect-all` for an existing persistent DC connection all we would need to provide is the `--device` option, which would simplify usage quite a bit.

martin-belanger, 2023-02-10:
For the `connect-all` command, I can confirm that when option `--device` is specified all other options are ignored ([fabrics.c#L888](https://github.com/linux-nvme/nvme-cli/blob/master/fabrics.c#L888)) and we simply retrieve the DLP from the existing DC connection.

For the I/O connections, however, the values specified on the command-line by options `--host-traddr` and `--host-iface` are used. If not specified, `--host-traddr` and `--host-iface` will simply default to `none` instead of defaulting to the values that were used for the DC connection ([fabrics.c#L879](https://github.com/linux-nvme/nvme-cli/blob/master/fabrics.c#L879)). I think this is a mistake.

When `--device` is used with `connect-all`, I'm proposing that we do as follows:
- If `--host-traddr` and/or `--host-iface` are specified, then we use those values for the I/O connections (no change with current behavior)
- If `--host-traddr` and/or `--host-iface` are not specified, then we default to those of the DC connection instead of `none` (this is the new behavior)

closed 2023-03-01 by igaw