linux-nvme/nvme-cli #1808: Reg firmware file size and 4k alignment

JohnGoogl, 2023-02-10:
It looks fw-download operation failing due to something related to firmware file size as for few cases its going OK.
So, please help me about firmware file size and fw-download scenarios which needs to take care and how nvme-cli handles same.
Actually, on same tried going through the documentation, but no much luck.

Also, looking for some insight about 4K alignment instead of other options, why this different alignments instead of universal/generic.

This queries might be dump, sorry, but am kind of new, so posted for clarifications pls, thanks in advance!

-John Jr.

igaw, 2023-02-10:
Do you happen to have some logs when things don't work? 

@jk-ozlabs

JohnGoogl, 2023-02-10:
Hello, @igaw 

I hope mention test logs helps.. let me know if any further details needed.

```
root@test:~# ls -l fw_*
-rwxr-xr-x 1 root root 2629632 Feb 11 01:01 fw_1.fluf
-rwxr-xr-x 1 root root 2629692 Feb 11 01:01 fw_2.fluf
-rwxr-xr-x 1 root root 2629637 Feb 11 01:02 fw_3.fluf
root@test:~# nvme fw-download /dev/nvme1 -f fw_1.fluf 
Firmware download success
root@test:~# nvme fw-download /dev/nvme1 -f fw_2.fluf 
NVMe status: INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(0x2)
root@test:~# nvme fw-download /dev/nvme1 -f fw_3.fluf 
Invalid size:2629637 for f/w image
root@test:~# 

```
Thanks!

igaw, 2023-02-13:
You can enable some more logs with providing the `--progress` command line option.

So let's have a look at the check conditions:

```c
	fw_size = sb.st_size;
	if ((fw_size & 0x3) || (fw_size == 0)) {
		fprintf(stderr, "Invalid size:%d for f/w image\n", fw_size);
		err = -EINVAL;
		goto close_fw_fd;
	}
```
As long the firmware is word aligned this check wont fire. fw2 seems to pass this check, while fw3 not.

The next step is

```c
	if (cfg.xfer == 0 || cfg.xfer % 4096)
		cfg.xfer = 4096;
```

Which means the transfer size is set to 4k if is not provided.

This is here is the main loop:
```c
	while (cfg.offset < fw_size) {
		cfg.xfer = min(cfg.xfer, fw_size);

		err = fw_download_single(dev, fw_buf + cfg.offset, fw_size,
					 cfg.offset, cfg.xfer, cfg.progress,
					 cfg.ignore_ovr);
		if (err)
			break;

		cfg.offset += cfg.xfer;
	}
```

`fw_download_single()` will call `nvme_cli_fw_download()` which seems to fail. I assume it is the final transfer when the transfer is `!= 4k`. You can verify it with the `--progress` command line option .

jk-ozlabs, 2023-02-13:
Note that some controllers will do odd things if you attempt to load a second firmware image without committing the first (or resetting the controller). Does changing the ordering change the behaviour here?

JohnGoogl, 2023-02-15:
@igaw and @jk-ozlabs thanks for the clarifications.

> Does changing the ordering change the behaviour here?
Yes, with order change also same behaviour.

Also, curious to know reasons for this alignment (4/8..k) and not sure either this process causing firmware file size increase to achieve this alignment cases, would appreciate your insights on same.

Thanks!

jk-ozlabs, 2023-02-15:
> Also, curious to know reasons for this alignment (4/8..k)

It's not a 4k alignment, it's a limit on the maximum command size. To transfer the firmware, we split it up into 4k blocks and issue those as separate Firmware Image Download commands that are small enough for the drive and/or transport.

The only "alignment" requirement is that the file size is a multiple of 4 bytes; that comes from the NVMe spec.

closed 2023-03-01 by igaw