linux-nvme/nvme-cli #1815: telemetry-log controller-initial log rae bit won't clear

xin2233, 2023-02-14:
According to the specification, when host get telemetry-log controller log done，telemetry Retain Asynchronous Event bit should clear. 
this is NVMe Spec 2.0b:
[image]

But in the code, the rae bit is '0' permanently. 
[image]

is it right ?
I think the rae bit should clear '0' when host get telemetry controller log done.

thanks.

igaw, 2023-02-15:
I did some history digging ended up here: 97ca8c0d1008 ("nvme-cli: add support of RAE"):

```
    nvme-cli: add support of RAE

    Currently telemetry-log cannot retain content of corresponding log page as RAE is not set.
    This patch adds RAE bit support into routines to retrieve log pages.
```

Since the adding the RAE bit it was set to true. These days we are using libnvme's `nvme_get_log_telemetry_ctrl()` which calls `nvme_get_log_page()` with following logic:

```c
int nvme_get_log_page(int fd, __u32 xfer_len, struct nvme_get_log_args *args)
{
	bool retain = true;

	do {

		/*
		 * Always retain regardless of the RAE parameter until the very
		 * last portion of this log page so the data remains latched
		 * during the fetch sequence.
		 */
		if (offset + xfer == data_len)
			retain = args->rae;
```

So the bare mechanics work fine (I just wanted to make sure we got that part right).

Anyway, it looks like the idea was to keep the log entry around and not clear it immediately. If I read the the portion of the spec correctly, it is not really violating the rules. It's not said you have to clear it. But yeah, it might be an option to make this optional via the command line. We got a bunch of other commands like the endurance event log where we allow to set/clear the rae flag from the command line.

igaw, 2023-02-24:
I've merged my PR. There was no feedback for one week. Let's see what I broke :)

closed 2023-02-24 by igaw

xin2233, 2023-02-24:
Hi igaw, 

I'm sorry I didn't reply in a week. I have read the nvme specification. My initial idea is that we should clean up to the available bit after exporting the telephone controller log, so that the device can continue to trigger the new telephone log. But I think your idea is reasonable.

thanks.

igaw, 2023-02-24:
I attempted not to break the existing behavior. But now you should be at least be able to overwrite it from the command line. Please give it a go. My quick test didn't show any problems.

PR #1823 (2023-02-24): telemetry-log refactoring and adding rae 
Allow to clear the rae bit for the telemetry-log command.

Fixes: #1815