linux-nvme/nvme-cli #1818: How to get rid of nvme-cli log size limit

kwin1412, 2023-02-15:
When I use nvme-cli(v2.2.1 and v2.3 ) and linux (5.15.0-60-generic) to debug NVMe SSD firmware, 

I send command 
`nvme get-log /dev/nvme0 -i 0x2 -l 0x200000`
 to get 2MB smart log.
But nvme-cli warns "Invalid arguments". 
And if I get less than 2MB log，it worked .
I checked my ssd firmware,it is  normal running. 

I located the limitation of the linux driver.
How to get rid of the log size limit?

igaw, 2023-02-15:
I've double checked if the parser of nvme-cli is not parsing the command line correctly. 

The 'invalid arguments' is an response from the kernel. `-i 0x2` is `SMART/Health Information Log Page` which is only defined up to 512 bytes length. `-l 0x200000` is just loo long. It should work fine with `-l 512`.

kwin1412, 2023-02-15:
In fact, I'm testing the vendor log page. It defines more than 0x200000 bytes length of structure.

If I want to get this vendor log page, I can not get this log page once.
Is there any way  to get rid of the log size limit on the kernel？

igaw, 2023-02-15:
I think the problem here is that we don't chunk the read. The `get-log` command uses the `nvme_get_log()` function which tries to read all in one go. Let me see if we couldn't do it with with `nvme_get_log_page()` which is able to chunk the read.

keithbusch, 2023-02-16:
I don't think the kernel will let you transfer that size in a single command.

closed 2023-02-16 by igaw

igaw, 2023-02-16:
The  AppImage for the current head:

  https://github.com/linux-nvme/nvme-cli/suites/11015832499/artifacts/558970880

With this version you should be able to fetch the data from your device. Notice the junk size is 4k which makes is really slow. You can experiment and use bigger chunks size to improve speed but as @keithbusch pointed out the kernel enforces some resource limits. One megabyte chunk size works on my machine, YMMV.

PR #1821 (2023-02-16): get-log: use chunked read
nvme: Use chunked read in get_log()

Large transfer might not work because we exhaust resource limits, so
chunk the read into 4k (default).

Fixes: #1818 
Depends: #https://github.com/linux-nvme/libnvme/pull/579