linux-nvme/nvme-cli #1827: tnvmcap and unvmcap in id-ctrl wrong data type and inconsistent formatting?

darren-c, 2023-02-16:
I noticed that in nvme-cli v2.3, the tnvmcap & unvmcap value in the id-ctrl query now includes commas.

nvme-cli 2.2.1
> tnvmcap   : 3200631791616

nvme-cli 2.3
> tnvmcap   : 3,200,631,791,616

I also noticed that if you use output-format=json, these values are set as strings (shouldn't they be numbers?)

>   "tnvmcap":"3,200,631,791,616",
     "unvmcap":"0",

I'm wondering if this is the intended behavior.

Of course it's mostly visual, however I had a Python script that was using these values and casting int() on them. However in nvme-cli v2.3, this causes an exception due to the commas. It's not a big deal to workaround it but thought it would make sense for nvme-cli to treat them as numbers. Thanks.

ikegami-t, 2023-02-16:
Yes the commas was added by the commit: 2a8bc94 ("nvme-print: Display smart log data units read and written SI value
"). About the json output changes I am not sure if we can fix so let me consult with @igaw . Probably it was changed for the review comment but seems to keep the previous behavior for JSON or change the localization as not available if needed to use. Do you have any comment or advice?
https://github.com/linux-nvme/nvme-cli/pull/1749#pullrequestreview-1234770082
> - drop uint128_t_to_string_tg, always use localization if available.

igaw, 2023-02-17:
I think adding the localization formatting to the JSON output was wrong (sorry.  We should revert it to plain numbers.

igaw, 2023-02-17:
This made me thinking we might want to write a schema. Though I don't know how we could easily test it. Without a regular test (CI) the schema is pretty worthless , IMO.

PR #1831 (2023-02-24): util: Revert JSON output to plain numbers
Add uint128_t_to_string_with_sep() to unset the thousands separator. Fixes: 2a8bc94 ("nvme-print: Display smart log data units read and written SI value") Fixes: #1827
closed 2023-02-24 by igaw