linux-nvme/nvme-cli #1841: Fabric connects come to a crawl when there are more than 50 existing connections

mhae, 2023-03-03:
I'm trying to connect to a couple hundred disk drives and we have multiple fabric paths to each of these drives. After ~50 connections, the time it takes to connect to a new paths is getting slower and slower. In our environment it can take an hour to connect to all paths.

I traced the fabrics connect code and the problem seems to be in the use of libnvme where we repeatedly go over all existing nvme sysfs entries. 
Here's a trace that shows the issue after we have ~50 nvme devices already connected:
```
// This trace message is printed at the beginning of nvme_scan_topology (libnvme:tree.c)
2023-03-03 14:44:54:293:nvme_scan_topology: enter
2023-03-03 14:44:54:293:after nvme_scan_ctrls
// Notice the time between nvme_scan_ctrlrs and "nvme_scan_ctrl loop". This is already 1s(!) and it gets worse the more ctrlrs we already have connected to.
2023-03-03 14:44:55:389:after nvme_scan_ctrl loop
2023-03-03 14:44:55:389:after nvme_scan_subsystems
2023-03-03 14:44:55:390:after nvme_scan_subsystem
2023-03-03 14:44:55:390:after  scan topo
2023-03-03 14:44:55:390:after lookup_ctrl
2023-03-03 14:44:55:390:after create
2023-03-03 14:44:55:644:after add
```

Here's the relevant code snippet (tree.c) that shows my trace messages (`m()`):
```
int nvme_scan_topology(struct nvme_root *r, nvme_scan_filter_t f, void *f_args)
{
	struct dirent **subsys, **ctrls;
	int i, num_subsys, num_ctrls, ret;

	if (!r)
		return 0;

	m("nvme_scan_topology: enter");
	num_ctrls = nvme_scan_ctrls(&ctrls);
	m("after nvme_scan_ctrls");
	if (num_ctrls < 0) {
		nvme_msg(r, LOG_DEBUG, "failed to scan ctrls: %s\n",
			 strerror(errno));
		return num_ctrls;
	}

	for (i = 0; i < num_ctrls; i++) {
		nvme_ctrl_t c = nvme_scan_ctrl(r, ctrls[i]->d_name);
		if (!c) {
			nvme_msg(r, LOG_DEBUG, "failed to scan ctrl %s: %s\n",
				 ctrls[i]->d_name, strerror(errno));
			continue;
		}
		m(ctrls[i]->d_name);

		if ((f) && !f(NULL, c, NULL, f_args)) {
			nvme_msg(r, LOG_DEBUG, "filter out controller %s\n",
				 ctrls[i]->d_name);
			nvme_free_ctrl(c);
		}
	}
	m("after nvme_scan_ctrl");
```

Note, an older version like 1.12 does not have this issue.

It seems the me we need the topology mainly for lookup_ctrlr to see if the ctrlr already exists. Is there a more efficient way to accomplish this?
Any other suggestion how to work around this issue?

igaw, 2023-03-07:
v1.x doesn't use the sysfs interface in this extend as the v2.x version does.

`nvme_scan_topology()` creates a tree structure which contains all controllers, subsystems and namespaces. All nvme fabric function iterator over this tree. It's not just for avoiding a error message. 

How do use the tool? Are you calling 50 times 'nvme-cli' or do you use `discover.conf` or `config.json`? In case you are calling 50 times nvme-cli, there isn't much we can do.

mhae, 2023-03-07:
Yes, I am be running "nvme connect ..." a ton of times. I've switched to "nvme connect-all" which significantly improves the connect time but I still have to call it 32 times, since we have 32 systems.
I can look at discover.conf but I'm not sure yet if this is something that will fit into what we're doing. 

I'm not familiar with config.json but I can give it a try...

Thanks.

igaw, 2023-03-08:
`config.json` is similiar to `discovery.conf`, a different format and you can configure more details.

Anyway, `nvme connect-all` is using a discovery controller to figure out what to do. So if your discovery controller contains all the information, a single `nvme connect-all` will do the trick. E.g. I've got a test setup with 50 namespaces and a signle `nvme connect-all` takes a few seconds to its job. (`nvme disconnect-all` is way slower but that is a known kernel limitation)

closed 2023-04-03 by igaw

mhae, 2023-04-03:
@martin-belanger , thanks for the heads up. Yeah, I actually wanted to try making connects in parallel but haven't had a chance. I guess I don't have to try that then...

I still think it would be worth the while to try to improve the nvme command when there are lots of already established connections.

igaw, 2023-04-04:
nvme-cli is a stateless tool, which means every time you start it, it reads the current state. Introducing any type of caching is going to open up a big door for all sorts of regressions. I really would like to avoid it. 

Though I am not saying there aren't code paths which can't be improved. Haven't really looked at it though, so I can't really comment on it.