linux-nvme/nvme-cli #1853: Could not nvme format --lbaf 1 an SKHynix HFS512GD9

infr-automation, 2023-03-11:
Could not nvme format --lbaf 1 an SKHynix HFS512GD9

even though the check showed that it had that lba format

nvme id-ns -H /dev/nvme0n1

This reformat is for the whole drive not for a partition, correct?

igaw, 2023-03-14:
Could you provide logs (shell and kernel logs) please?

infr-automation, 2023-03-16:
[syslog1.zip](https://github.com/linux-nvme/nvme-cli/files/10993021/syslog1.zip)

keithbusch, 2023-03-16:
Could you try adding param "`--namespace-id=0xffffffff`"?

keithbusch, 2023-03-16:
Ok, sounds like a vendor firmware bug. You're doing everything correctly, but the drive is reporting an error anyway.

infr-automation, 2023-03-16:
nvme list
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            NJ04N638012104E57    SKHynix_HFS512GD9TNG-L5B0B               1         512.11  GB / 512.11  GB    512   B +  0 B   80750C10

I have not found newer yet.

infr-automation, 2023-03-16:
nvme list
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            NJ04N638012104E57    SKHynix_HFS512GD9TNG-L5B0B               1         512.11  GB / 512.11  GB    512   B +  0 B   80750C10

I have not found newer yet.

https://github.com/linux-nvme/nvme-cli/issues/816 here they do a suspend for workaround

closed 2023-03-17 by igaw