linux-nvme/nvme-cli #1879: build error

saybeking, 2023-03-28:
ninja: Entering directory `.build'
[18/74] Compiling C object nvme.p/nvme.c.o
../nvme.c: In function ‘gen_tls_key’:
../nvme.c:8933:13: warning: implicit declaration of function ‘nvme_insert_tls_key’ [-Wimplicit-function-declaration]
 8933 |   tls_key = nvme_insert_tls_key(cfg.keyring, cfg.keytype,
      |             ^~~~~~~~~~~~~~~~~~~
[72/74] Compiling C object nvme.p/fabrics.c.o
FAILED: nvme.p/fabrics.c.o 
ccache cc -Invme.p -I. -I.. -Iccan -I../ccan -I/usr/local/include -I/usr/include/json-c -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -g -fomit-frame-pointer -D_GNU_SOURCE -include config.h -DCCAN_LIST_DEBUG=1 -DCCAN_STR_DEBUG=1 -MD -MQ nvme.p/fabrics.c.o -MF nvme.p/fabrics.c.o.d -o nvme.p/fabrics.c.o -c ../fabrics.c
In file included from ../nvme.h:31,
                 from ../fabrics.c:42:
../fabrics.c: In function ‘discover_from_conf_file’:
../fabrics.c:105:39: error: ‘struct nvme_fabrics_config’ has no member named ‘keyring’
  105 |  OPT_INT("keyring",              0, &c.keyring,            nvmf_keyring), \
      |                                       ^
../util/argconfig.h:80:25: note: in definition of macro ‘OPT_INT’
   80 |  {l, s, "NUM", CFG_INT, v, required_argument, d}
      |                         ^
../fabrics.c:537:3: note: in expansion of macro ‘NVMF_OPTS’
  537 |   NVMF_OPTS(cfg),
      |   ^~~~~~~~~
../fabrics.c:106:39: error: ‘struct nvme_fabrics_config’ has no member named ‘tls_key’
  106 |  OPT_INT("tls_key",              0, &c.tls_key,            nvmf_tls_key), \
      |                                       ^
../util/argconfig.h:80:25: note: in definition of macro ‘OPT_INT’
   80 |  {l, s, "NUM", CFG_INT, v, required_argument, d}
      |                         ^
../fabrics.c:537:3: note: in expansion of macro ‘NVMF_OPTS’
  537 |   NVMF_OPTS(cfg),
      |   ^~~~~~~~~
../fabrics.c: In function ‘nvmf_discover’:
../fabrics.c:105:39: error: ‘struct nvme_fabrics_config’ has no member named ‘keyring’
  105 |  OPT_INT("keyring",              0, &c.keyring,            nvmf_keyring), \
      |                                       ^
../util/argconfig.h:80:25: note: in definition of macro ‘OPT_INT’
   80 |  {l, s, "NUM", CFG_INT, v, required_argument, d}
      |                         ^
../fabrics.c:720:3: note: in expansion of macro ‘NVMF_OPTS’
  720 |   NVMF_OPTS(cfg),
      |   ^~~~~~~~~
../fabrics.c:106:39: error: ‘struct nvme_fabrics_config’ has no member named ‘tls_key’
  106 |  OPT_INT("tls_key",              0, &c.tls_key,            nvmf_tls_key), \
      |                                       ^
../util/argconfig.h:80:25: note: in definition of macro ‘OPT_INT’
   80 |  {l, s, "NUM", CFG_INT, v, required_argument, d}
      |                         ^
../fabrics.c:720:3: note: in expansion of macro ‘NVMF_OPTS’
  720 |   NVMF_OPTS(cfg),
      |   ^~~~~~~~~
../fabrics.c: In function ‘nvmf_connect’:
../fabrics.c:105:39: error: ‘struct nvme_fabrics_config’ has no member named ‘keyring’
  105 |  OPT_INT("keyring",              0, &c.keyring,            nvmf_keyring), \
      |                                       ^
../util/argconfig.h:80:25: note: in definition of macro ‘OPT_INT’
   80 |  {l, s, "NUM", CFG_INT, v, required_argument, d}
      |                         ^
../fabrics.c:908:3: note: in expansion of macro ‘NVMF_OPTS’
  908 |   NVMF_OPTS(cfg),
      |   ^~~~~~~~~
../fabrics.c:106:39: error: ‘struct nvme_fabrics_config’ has no member named ‘tls_key’
  106 |  OPT_INT("tls_key",              0, &c.tls_key,            nvmf_tls_key), \
      |                                       ^
../util/argconfig.h:80:25: note: in definition of macro ‘OPT_INT’
   80 |  {l, s, "NUM", CFG_INT, v, required_argument, d}
      |                         ^
../fabrics.c:908:3: note: in expansion of macro ‘NVMF_OPTS’
  908 |   NVMF_OPTS(cfg),
      |   ^~~~~~~~~
../fabrics.c: In function ‘nvmf_config’:
../fabrics.c:105:39: error: ‘struct nvme_fabrics_config’ has no member named ‘keyring’
  105 |  OPT_INT("keyring",              0, &c.keyring,            nvmf_keyring), \
      |                                       ^
../util/argconfig.h:80:25: note: in definition of macro ‘OPT_INT’
   80 |  {l, s, "NUM", CFG_INT, v, required_argument, d}
      |                         ^
../fabrics.c:1223:3: note: in expansion of macro ‘NVMF_OPTS’
 1223 |   NVMF_OPTS(cfg),
      |   ^~~~~~~~~
../fabrics.c:106:39: error: ‘struct nvme_fabrics_config’ has no member named ‘tls_key’
  106 |  OPT_INT("tls_key",              0, &c.tls_key,            nvmf_tls_key), \
      |                                       ^
../util/argconfig.h:80:25: note: in definition of macro ‘OPT_INT’
   80 |  {l, s, "NUM", CFG_INT, v, required_argument, d}
      |                         ^
../fabrics.c:1223:3: note: in expansion of macro ‘NVMF_OPTS’
 1223 |   NVMF_OPTS(cfg),
      |   ^~~~~~~~~
[73/74] Compiling C object nvme.p/nvme-print.c.o
FAILED: nvme.p/nvme-print.c.o 
ccache cc -Invme.p -I. -I.. -Iccan -I../ccan -I/usr/local/include -I/usr/include/json-c -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -g -fomit-frame-pointer -D_GNU_SOURCE -include config.h -DCCAN_LIST_DEBUG=1 -DCCAN_STR_DEBUG=1 -MD -MQ nvme.p/nvme-print.c.o -MF nvme.p/nvme-print.c.o.d -o nvme.p/nvme-print.c.o -c ../nvme-print.c
../nvme-print.c: In function ‘nvme_show_error_log’:
../nvme-print.c:3801:36: error: ‘struct nvme_error_log_page’ has no member named ‘csi’; did you mean ‘cs’?
 3801 |   printf("csi  : %d\n", err_log[i].csi);
      |                                    ^~~
      |                                    cs
../nvme-print.c:3802:39: error: ‘struct nvme_error_log_page’ has no member named ‘opcode’
 3802 |   printf("opcode  : %#x\n", err_log[i].opcode);
      |                                       ^
../nvme-print.c:3806:46: error: ‘struct nvme_error_log_page’ has no member named ‘log_page_version’
 3806 |   printf("log_page_version: %d\n", err_log[i].log_page_version);
      |                                              ^
ninja: build stopped: subcommand failed.

minwooim, 2023-03-28:
Please update the `libnvme` installed in your machine to have the following commit:

https://github.com/linux-nvme/libnvme/commit/629b04525a5d5d076db2908c533324bd910ec6c6

igaw, 2023-03-28:
`meson subprojects update` should take care of it assuming you are not building against a locally install libnvme. If so you should tell nvme-cli to use the fallback with `meson setup .build --force-fallback-for=libnvme`

closed 2023-03-28 by igaw