linux-nvme/nvme-cli #1880: Date and Time of "Error Statistics" for Smart Log of an SSD

lamyergeier, 2023-03-29:
- The following command returns immediately, indicating that the test for errors are not performed in the realtime, as extended long test would need some time. How to know when was the last test performed and how often the test is done? Also, is the test detailed and can it be trusted to indicate the health of SSD?

```bash
$ sudo nvme smart-log /dev/nvme0n1
Smart Log for NVME device:nvme0n1 namespace-id:ffffffff
critical_warning                        : 0
temperature                             : 39 C (312 Kelvin)
available_spare                         : 100%
available_spare_threshold               : 10%
percentage_used                         : 0%
endurance group critical warning summary: 0
data_units_read                         : 8,277,276
data_units_written                      : 6,579,167
host_read_commands                      : 78,760,319
host_write_commands                     : 120,191,689
controller_busy_time                    : 171
power_cycles                            : 202
power_on_hours                          : 173
unsafe_shutdowns                        : 44
media_errors                            : 0
num_err_log_entries                     : 0
Warning Temperature Time                : 0
Critical Composite Temperature Time     : 0
Temperature Sensor 1           : 51 C (324 Kelvin)
Temperature Sensor 2           : 39 C (312 Kelvin)
Thermal Management T1 Trans Count       : 0
Thermal Management T2 Trans Count       : 0
Thermal Management T1 Total Time        : 0
Thermal Management T2 Total Time        : 0
```

keithbusch, 2023-03-29:
Are you looking for the `device-self-test` command instead? The `smart-log` is just a live snapshot of running event counters and sensors.

igaw, 2023-04-03:
No feedback. Closing.

lamyergeier, 2023-04-03:
@igaw Sorry for the delay.

I get the following output:

```bash
$ sudo nvme device-self-test /dev/nvme0n1
NVMe status: INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(0x4002)
```

I checked this: [[INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(0x4002) · Issue #1130 · linux-nvme/nvme-cli](https://github.com/linux-nvme/nvme-cli/issues/1130)](https://github.com/linux-nvme/nvme-cli/issues/1130)

Does it mean that device self test is not supported?

keithbusch, 2023-04-03:
I would expect the device to return INVALID_OPCODE error if it didn't support device self test. But this error code you see doesn't sound valid if the device was implementing this command.

lamyergeier, 2023-04-03:
@keithbusch  Any suggestions, on what I could do? Any commands to investigate this further?

keithbusch, 2023-04-03:
Try `nvme id-ctrl /dev/nvme0 -H | grep "Device Self"` and see if that says if it's supported.

lamyergeier, 2023-04-03:
```bash
$ sudo nvme id-ctrl /dev/nvme0 -H | grep "Device Self"
Place your right index finger on the fingerprint reader
  [4:4] : 0x1   Device Self-test Supported
```

keithbusch, 2023-04-04:
Sorry, I forgot to mention you need to specify the self test code in order to initiate it. The default code isn't necessarily valid. So try `nvme device-self-test /dev/nvme0 -s 1` to start a "short" self test. When it completes, which happens asynchronously, you can check the results with `nvme self-test-log /dev/nvme0`.

lamyergeier, 2023-04-04:
@keithbusch I completed short and extended test. 

```bash
$ sudo nvme self-test-log /dev/nvme0 -v
Device Self Test Log for NVME device:nvme0
Current operation  : 0
Current Completion : 0%
Self Test Result[0]:
  Operation Result             : 0 Operation completed without error
  Self Test Code               : 2 Extended device self-test operation
  Valid Diagnostic Information : 0
  Power on hours (POH)         : 0xb8
  Vendor Specific              : 0 0
Self Test Result[1]:
  Operation Result             : 0 Operation completed without error
  Self Test Code               : 1 Short device self-test operation
  Valid Diagnostic Information : 0
  Power on hours (POH)         : 0xb7
  Vendor Specific              : 0 0
```

Both of these test show `Operation Result             : 0 Operation completed without error`, does that mean no test were successful (because it implies that some operations were completed with error!)? 

Is there any thing useful in the output of the test log? Are there any other command to see detailed output?

keithbusch, 2023-04-04:
I'm not sure I understand your question. The text says "without error", so where are you finding info that implies otherwise?

The log says everything is okay from the device's perspective; the device is saying it completed it's diagnostics and didn't find anything to report. Nothing else interesting here. What more details are you looking for?

lamyergeier, 2023-04-04:
It says `0 Operation completed without error`, that means not even a single operation completed without error. So, either none of the operation got completed. Or if some were completed they ended up with error.

keithbusch, 2023-04-04:
Oh, maybe we can improve the output since that might be confusing. This utility first prints the raw self test status code, `0`, then it prints a literal copy-paste from the spec of what `0` means (it's on NVMe Spec page 191, figure 213).

```
0h  |  Operation completed without error
```

lamyergeier, 2023-04-04:
So, if the status is 0 as shown above it indicates that none of the following have occurred except 0h

[image]

That means that SSD is definitely in a good health. Is it?

keithbusch, 2023-04-04:
`0` means success, no errors. That's all it means.

PoH is supposed to be power on hours since the device was manufactured. There's no spec way to reset it.

It seems like you badly want this utility to report something negative about your device. Whether or not this self-test operation and various logs are useful really depends on your implementation. The spec just defines the command and response formats, and this tool is just the messenger. The extent to which the device faithfully implements it, and trustworthiness of the output is all vendor/model specific.

While unlikely, there are reasons other than an imminent device failure that could cause a filesystem to become read-only. If there really were SSD read/write errors, there should definitely be evidence of that in your kernel's `dmesg` around the point it becomes read-only, which is probably where we should have started.

igaw, 2023-04-05:
> Oh, maybe we can improve the output since that might be confusing. This utility first prints the raw self test status code, `0`, then it prints a literal copy-paste from the spec of what `0` means (it's on NVMe Spec page 191, figure 213).

We do this type of value return in a lot of places. The only real difference between, let's say `id-ctrl` and this self test is in the `id-ctrl` output we have a couple more spaces between the value and the help text. I suggest we do the same here or change all other places along side. My OCD wants it to be consistent :)