linux-nvme/nvme-cli #1893: nvme-fc autoconnect scripts fails to parse config JSON file

martin-gpy, 2023-04-04:
There is currently no logic in the nvme-fc autoconnect scripts to parse the config JSON file and connect accordingly.

The below patch is available today, but doesn't serve the purpose:

fabrics: Trigger auto connect if config.json exists - https://github.com/linux-nvme/nvme-cli/pull/1694/commits/eec7634bbb17e1604c15ab47f46f44a4d524a043

igaw, 2023-04-04:
The `ConditionPathExists=|!` is supposed work as `OR` condition and `nvme connect-all` is checking both files. Do you have some logs for it? Is the unit never triggered? If so it might be another instance of boot ordering.

https://github.com/linux-nvme/nvme-cli/blob/a298e39fcb668dd0e814d49c373a358734aedf5e/nvmf-autoconnect/systemd/nvmf-autoconnect.service.in#L5-L6

- https://www.freedesktop.org/software/systemd/man/systemd.unit.html#Conditions%20and%20Asserts
- https://stackoverflow.com/questions/37313677/what-is-the-difference-between-conditionpathexists-and-conditionpathexists-in
- https://www.freedesktop.org/software/systemd/man/bootup.html

igaw, 2023-05-24:
After reading the upstream documentation on this topic:

  https://www.freedesktop.org/software/systemd/man/systemd.unit.html

I see where I got fooled by the StackOverflow article. The 'OR' part is just `Condition...=|` and not `Condition...=|!`.

martin-gpy, 2023-05-29:
This issue is addressed with the following:

1) fabrics: fix conditions in nvmf-autoconnect.service - https://github.com/linux-nvme/nvme-cli/commit/9803d92f073a4c8c398de05e6edb408565d8faf9

2) Enabling nvmf-autoconnect.service to start at bootup

We can go ahead and close this issue now.

igaw, 2023-06-01:
Thanks for the feedback. 

BTW, one thing I which starts to bother me is the inconsistent naming of the systemd service files. Also no documentation whatsoever how to use them.

closed 2023-06-01 by igaw