linux-nvme/nvme-cli #1897: "failed to set block size to 1" but seems to have worked

dreadedhamish, 2023-04-09:
Following recommendations from the ZFS docs I was trying to reformat my nvme drive with a block size of 4096.
Using smartctl I could see it was set to the less performant 512:

> Supported LBA Sizes (NSID 0x1)
> Id Fmt  Data  Metadt  Rel_Perf
>  0 +     512       0         1
>  1 -    4096       0         0

I used the following commend to reformat using ID=1:

> sudo nvme format /dev/nvme1n1 -l 1

And received this mixed message:

> Success formatting namespace:ffffffff
> failed to set block size to 1

Running smartctl again it looks like it was successful:
Supported LBA Sizes (NSID 0x1)

> Id Fmt  Data  Metadt  Rel_Perf
>  0 -     512       0         1
>  1 +    4096       0         0

Should I assume it was ultimately successful or not?

nvme-cli: 1.16
ubuntu: 22.10
drive: crucial p3 2TB
model: CT2000P3SSD8
firmware: P9CR30A
PCI Vendor/Subsystem ID: 0xc0a9
IEEE OUI Identifier: 0x00a075

ikegami-t, 2023-04-11:
It seems that the error is depended on the kernel version or drive behavior since it is not caused on my local environment.
```
tokunori@tokunori-desktop:~/nvme-cli$ sudo nvme format /dev/nvme0n1 -l 1
You are about to format nvme0n1, namespace 0x1.
Namespace nvme0n1 has parent controller(s):nvme0

WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force|-f] option to suppress this warning.
Sending format operation ... 
Success formatting namespace:1
tokunori@tokunori-desktop:~/nvme-cli$ sudo nvme format /dev/nvme1n1 -l 1
You are about to format nvme1n1, namespace 0x1.
Namespace nvme1n1 has parent controller(s):nvme1

WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force|-f] option to suppress this warning.
Sending format operation ... 
Success formatting namespace:1
tokunori@tokunori-desktop:~/nvme-cli$ nvme version
nvme version 1.16
tokunori@tokunori-desktop:~/nvme-cli$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.10
DISTRIB_CODENAME=kinetic
DISTRIB_DESCRIPTION="Ubuntu 22.10"
tokunori@tokunori-desktop:~/nvme-cli$ uname -a
Linux tokunori-desktop 6.2.0+ #1 SMP PREEMPT_DYNAMIC Sun Feb 26 23:27:57 JST 2023 x86_64 x86_64 x86_64 GNU/Linux
tokunori@tokunori-desktop:~/nvme-cli$ sudo nvme list
Node                  SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          S63WNE0R100201       SAMSUNG MZQL2960HCJR-00B7C               1           0.00   B / 960.20  GB      4 KiB +  0 B   GDA53C2Q
/dev/nvme1n1          50026B76862F98CA     KINGSTON SEDC1000BM8240G                 1         240.06  GB / 240.06  GB    512   B +  0 B   ECEK22.3
```

dreadedhamish, 2023-04-12:
But it's an error thrown by nvme-cli, yeah?
What would cause it to throw such an error, and how might it throw an error while being successful?

igaw, 2023-04-12:
Could you retest with latest upstream version of nvme-cli (and maybe even the kernel)?

ikegami-t, 2023-04-12:
The issue not caused on the older version kernel also so seems not depended on the kernel version also but only depended on the drive.
```sh
tokunori@tokunori-desktop:~/nvme-cli$ uname -a
Linux tokunori-desktop 5.19.0-38-generic #39-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 17 17:33:16 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
tokunori@tokunori-desktop:~/nvme-cli$ sudo nvme list
Node                  SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          S63WNE0R100201       SAMSUNG MZQL2960HCJR-00B7C               1           0.00   B / 960.20  GB      4 KiB +  0 B   GDA53C2Q
tokunori@tokunori-desktop:~/nvme-cli$ sudo nvme format /dev/nvme0n1 -l 1
You are about to format nvme0n1, namespace 0x1.
Namespace nvme0n1 has parent controller(s):nvme0

WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force|-f] option to suppress this warning.
Sending format operation ... 
Success formatting namespace:1
```

ikegami-t, 2023-04-12:
The error is output by the following format command code part.
```c
static int format(int argc, char **argv, struct command *cmd, struct plugin *plugin)
{
...
				block_size = 1 << ns.lbaf[cfg.lbaf].ds;

				/*
				 * If block size has been changed by the format
				 * command up there, we should notify it to
				 * kernel blkdev to update its own block size
				 * to the given one because blkdev will not
				 * update by itself without re-opening fd.
				 */
				if (ioctl(fd, BLKBSZSET, &block_size) < 0) {
					fprintf(stderr, "failed to set block size to %d\n",
							block_size);
					err = -errno;
					goto close_fd;
				}
```

igaw, 2023-04-12:
Oohh, that rings a bell. 

https://lore.kernel.org/linux-nvme/CADqNVToVWkJHSPbYqJRgnFjdMt3N_-UkCHZtHVTRhpGNThX3Wg@mail.gmail.com/#t

IIRC, @keithbusch also mentioned that we might just remove the ioctl...

ikegami-t, 2023-04-12:
On my local environment still the NSID 0xffffffff also not failed to format.
```sh
tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme format /dev/nvme0 -l 1 -j -n 0xffffffff
You are about to format nvme0, namespace 0xffffffff(ALL namespaces).
WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force] option to suppress this warning.
Sending format operation ...
Success formatting namespace:ffffffff
tokunori@tokunori-desktop:~/nvme-cli$ sudo nvme format /dev/nvme0 -l 1 -n 0xffffffff
You are about to format nvme0, namespace 0xffffffff(ALL namespaces).
Controller nvme0 has child namespace(s):nvme0n1

WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force|-f] option to suppress this warning.
Sending format operation ...
Success formatting namespace:ffffffff
tokunori@tokunori-desktop:~/nvme-cli$ sudo nvme format /dev/nvme1 -l 1 -n 0xffffffff
You are about to format nvme1, namespace 0xffffffff(ALL namespaces).
Controller nvme1 has child namespace(s):nvme1n1

WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force|-f] option to suppress this warning.
Sending format operation ...
Success formatting namespace:ffffffff
tokunori@tokunori-desktop:~/nvme-cli$ sudo nvme list
Node                  SN                   Model                                    Namespace Usage                      Format           FW Rev
--------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          S63WNE0R100201       SAMSUNG MZQL2960HCJR-00B7C               1           0.00   B / 960.20  GB      4 KiB +  0 B   GDA53C2Q
/dev/nvme1n1          50026B76862F98CA     KINGSTON SEDC1000BM8240G                 1         240.06  GB / 240.06  GB      4 KiB +  0 B   ECEK22.3
```

ikegami-t, 2023-04-12:
The error is returned by the following kernel code functions.
```c
static int blkdev_bszset(struct block_device *bdev, fmode_t mode,
		int __user *argp)
{
	int ret, n;

	if (!capable(CAP_SYS_ADMIN))
		return -EACCES;
	if (!argp)
		return -EINVAL;
	if (get_user(n, argp))
		return -EFAULT;

	if (mode & FMODE_EXCL)
		return set_blocksize(bdev, n);

	if (IS_ERR(blkdev_get_by_dev(bdev->bd_dev, mode | FMODE_EXCL, &bdev)))
		return -EBUSY;
	ret = set_blocksize(bdev, n);
	blkdev_put(bdev, mode | FMODE_EXCL);

	return ret;
}

int set_blocksize(struct block_device *bdev, int size)
{
	/* Size must be a power of two, and between 512 and PAGE_SIZE */
	if (size > PAGE_SIZE || size < 512 || !is_power_of_2(size))
		return -EINVAL;

	/* Size cannot be smaller than the size supported by the device */
	if (size < bdev_logical_block_size(bdev))
		return -EINVAL;

	/* Don't change the size if it is same as current */
	if (bdev->bd_inode->i_blkbits != blksize_bits(size)) {
		sync_blockdev(bdev);
		bdev->bd_inode->i_blkbits = blksize_bits(size);
		kill_bdev(bdev);
	}
	return 0;
}
```

ikegami-t, 2023-04-12:
Just reproduced the issue as below as specifies the block device and NSID 0xffff and if specifies the character device or NSID 1 the issue not happened.
```sh
tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme format /dev/nvme0n1 -l 1 -n 0xffffffff
You are about to format nvme0n1, namespace 0xffffffff(ALL namespaces).
WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force] option to suppress this warning.
Sending format operation ... 
Success formatting namespace:ffffffff
failed to set block size to 1
tokunori@tokunori-desktop:~/nvme-cli$ sudo .build/nvme format /dev/nvme0n1 -l 1
You are about to format nvme0n1, namespace 0x1.
WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force] option to suppress this warning.
Sending format operation ... 
Success formatting namespace:1
```

igaw, 2023-04-12:
`nvme_update_ns_info()` needs to be executed before the `ioctl` happens. And this will happen via the scan algorithm. Hence if the ioctl is too fast it will fail (updating `logic_block_size` via `blk_queue_logical_block_size()` ).

igaw, 2023-04-12:
If you add a sleep between format and ioctl it should be working more often

dreadedhamish, 2023-04-13:
Off all the tickets I've filed for bugs, this has been the best response.

igaw, 2023-04-13:
> Just created PR #1900 to fix this issue so review it.

Nice work! I've applied it. 

Let's close this report, if there is still a problem, please just reopen it and I am sure @ikegami-t will fix the problem! :)

closed 2023-04-13 by igaw