linux-nvme/nvme-cli #1903: nvme crashes on run, and gives random values as output

antonyh, 2023-04-13:
nvme crashes on run, and gives random values as output.

example, every run gives a different NVMe specification value, and ends with double free or corruption.

Running Debian 12 aka Bookworm, using a Crucial P3 CT4000P3SSD8. I can update with mainboard details or other information if useful to diagnose this problem.
```
$ apt info nvme-cli
Package: nvme-cli
Version: 2.3-2

$ sudo nvme --version
nvme version 2.3 (git 2.3)
libnvme version 1.3 (git 1.3)

$ sudo nvme show-regs -H /dev/nvme0n1
get-property: Invalid argument
cap     : 55a793da2290
Controller Ready With Media Support (CRWMS): Not Supported
Controller Ready Independent of Media Support (CRIMS): Not Supported
Controller Memory Buffer Supported (CMBS): The Controller Memory Buffer is Not Supported
Persistent Memory Region Supported (PMRS): The Persistent Memory Region is Not Supported
Memory Page Size Maximum         (MPSMAX): 4096 bytes
Memory Page Size Minimum         (MPSMIN): 4096 bytes
Boot Partition Support              (BPS): No
Command Sets Supported              (CSS): NVM command set is Supported
One or more I/O Command Sets are Not Supported
Only Admin Command Set Supported
NVM Subsystem Reset Supported     (NSSRS): No
Doorbell Stride                   (DSTRD): 512 bytes
Timeout                              (TO): 73500 ms
Arbitration Mechanism Supported     (AMS): Weighted Round Robin with Urgent Priority Class is supported
Contiguous Queues Required          (CQR): No
Maximum Queue Entries Supported    (MQES): 8849

version : d1f52cc0
NVMe specification 53749.44

cc      : 0
Controller Ready Independent of Media Enable (CRIME): Disabled
I/O Completion Queue Entry Size (IOCQES): 1 bytes
I/O Submission Queue Entry Size (IOSQES): 1 bytes
Shutdown Notification              (SHN): No notification; no effect
Arbitration Mechanism Selected     (AMS): Round Robin
Memory Page Size                   (MPS): 4096 bytes
I/O Command Set Selected           (CSS): NVM Command Set
Enable                              (EN): No

csts    : 0
Processing Paused               (PP): No
NVM Subsystem Reset Occurred (NSSRO): No
Shutdown Status               (SHST): Normal operation (no shutdown has been requested)
Controller Fatal Status        (CFS): False
Ready                          (RDY): No

double free or corruption (!prev)
Aborted
```

ikegami-t, 2023-04-15:
The crash itself has been fixed by the PR #1845 changes c5099ec4a but still not sure about the random values cause.

ikegami-t, 2023-04-15:
Sorry the random values also caused by the same cause fixed by the PR so the issue behaiors crash and random values fixed by the latest nvme-cli.
Note: The following error message behavior only seems still not fixed by the PR changes.
```sh
$ sudo nvme show-regs -H /dev/nvme0n1
get-property: Invalid argument
```

igaw, 2023-04-17:
@ikegami-t thanks for the analysis. 

Not much we can do about the downstream version. In case you want to test the latest greatest, we provide a precompiled version as `AppImage`: https://monom.org/linux-nvme/upload/AppImage/nvme-cli-latest-x86_64.AppImage

Anyway, let's close the bug report.

antonyh, 2023-04-17:
Thanks @ikegami-t for the update, it sounds like this has already been mostly fixed upstream and I just need to wait for it to flow into my distro.